GoogleCloudPlatform/google-cloud-iot-arduino #148: publishState causes disconnection

SeanBurford, 2020-01-23:
If you use the cloud console to send a config to a device, it will receive a message on /config when it next reconnects.  Responding to this message with a publishState will cause disconnection (and the same will happen again on every reconnection).

Receiving configuration and then publishing updated state is documented here:
https://cloud.google.com/iot/docs/concepts/devices

This disconnection is caused by CloudIoTCoreMqtt.cpp CloudIoTCoreMqtt::onConnect() calling publishState("connected").  Just after onConnect(), Google is going to publish the latest config to /config and the device may want to reply with publishState() to acknowledge the config and update the Google-side stored state.  As shown, acknowledging like this will cause an immediate disconnection.

Note that "connected" is a bit of a silly state for the library to publish as it will result in Cloud Console thinking that the device has acknowledged any config that was previously published to the device.

gguuss, 2020-02-06:
Are you saying that [this code](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/src/CloudIoTCoreMqtt.cpp#L295) is causing a disconnect? Which board / example are you using? You could [disable this with the **void CloudIoTCoreMqtt::setLogConnect** helper](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/src/CloudIoTCoreMqtt.cpp#L300) as a temporary work around, I have seen issues with connection timing on the ESP8266 and network calls on the lwMQTT library maybe there's something going on with too many network calls causing a WDT reset.

explowable, 2020-03-24:
Same Issue here. In my case only [this publishState](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/src/CloudIoTCoreMqtt.cpp#L295) is working.

After other publishState the device is crashing/disconnecting. 😓 

**Sample Code:**
```
// The MQTT callback function for commands and configuration updates
void messageReceived(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload);
  
  if (topic.lastIndexOf(configTopic) > 0) {
    configUpdate(payload);
    publishState(String(payload));
  }
  if (topic.lastIndexOf(commandTopic) > 0) {
    commandUpdate(payload);
    publishState(String(payload));
  }
}
```

Here I am trying to send a "verification" state update after an incoming configuration update. Maybe I'm doing something wrong? If so, could you please tell me the right way? Thank you! @gguuss

gguuss, 2020-03-25:
Oof. I see what's goin on now. I am guessing the ESP is having issues context switching between the publish and receive operations. Another potential source of issue is that due to the "at least once delivery" nature of PubSub messages the callback could get triggered multiple times before the state messages are sent. You are limited to transmitting one state message per second and this potentially could cause a disconnect (but I don't think it does).

I would recommend the following:

* Create a queue of state messages
* Append to the queue in the callback
* Dequeue the state message[s] and transmit them outside of the callback.

I'm context switching a bit between this and other work and the pandemic is rate-limiting me a bit but I've added an issue to put this functionality into the library.

explowable, 2020-03-25:
> Oof. I see what's goin on now. I am guessing the ESP is having issues context switching between the publish and receive operations. Another potential source of issue is that due to the "at least [...]

Thank you very much for your fast answer! I was thinking about the limitation of the state messages too, but couldn't figure out a solution. 

> I would recommend the following:
> 
> * Create a queue of state messages
> * Append to the queue in the callback
> * Dequeue the state message[s] and transmit them outside of the callback.

I will try that, thank you very much. Hope I'm skilled enough. 😓

galz10, 2020-07-01:
[Wiki Page](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/wiki/PublishState-Causes-Disconnection)

closed 2020-07-01 by galz10

thilojaeggi, 2021-07-11:
Did anyone find a solution to this? I don't think I'm skilled enough.