GoogleCloudPlatform/google-cloud-iot-arduino #149: Simcom SIM7020E MQTT Password length is limited

tortechnocom, 2020-02-06:
I have try to use Simcom SIM7020 that MQTT password limit (Max 100 chars) on page 215.

*** Do your developers have another solutions or suggestions?

http://microchip.ua/simcom/LTE/SIM7020/SIM7020%20Series_AT%20Command%20Manual_V1.03.pdf

gguuss, 2020-02-06:
It's going to be the JWT which does get pretty long. I think that anecdotally ES256 JWTs will be shorter than RS256 but I think you would need to find a board work around for supporting a longer password. You could alternatively [use a gateway](https://cloud.google.com/iot/docs/how-tos/gateways/mqtt-bridge) to bridge the connection from these devices - I would advise that you have a separate auth for bridging this if your network cannot be secured.

gguuss, 2020-10-15:
This is a duplicate of https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/142

closed 2020-10-15 by gguuss