GoogleCloudPlatform/google-cloud-iot-arduino #152: Library weight

NicolasLiveris, 2020-03-02:
Hi there,

I have a compilation problem when I use the library. It seems to be too big for the arduino UNO. 

[image]

And when i comment this line:
[image]

Everything is ok:
[image]

However, this one works with an ESP8266 which doesn't take much more memory. How can I reduce the size of the library?

Thank you, and have a nice day.

MrEditor97, 2020-03-02:
Hi,
The library currently only supports these main platforms:
- Genuino MKR1000 and WiFi101
- Espressif ESP32
- Espressif ESP8266

As can be seen here: @[Supported Hardware](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/README.md#supported-hardware-targets) 

The library won't work on a Ardunio UNO even if you managed to fit it onto the board itself, the UNO doesn't have WiFi or Ethernet so the library would be pointless.
Hope this helps.

NicolasLiveris, 2020-03-02:
hi @MrEditor97

I didn't explain myself well enough. I'm trying to run the library on an arduino Uno with an ethernet shield. The basic goal was to just test the library on Uno. However, after adapting for the ethernet interface, the only problem I can't solve is the memory space. This will require more code rewriting to save space. Hoping to get there.

Thank you for answering.

gguuss, 2020-03-24:
The UNO has 32kb of memory whereas the ESP has either 1024kb or 4096kb, a difference of 30-100x. I'm not sure reducing the size of the library (even just the JWT library) would be feasible (the initial contributor was targeting the ATmega328P and was struggling both with memory/cpu limitations). The MKR board shares some library capabilities with the Uno but the network interface would still require porting to make this work.

The supported boards are fairly inexpensive and easy to get - is there a specific reason you want to use the Uno and cannot use one of the supported boards?

cirvladimir, 2020-04-24:
Sorry for the late addition, but I think it might be doable (although it would be a challenge). I believe I was able to do an ecdsa sign on the Arduino, which should be the heaviest part. The token itself should fit in RAM. Then only other issue is the server might use RSA for HTTPS/TLS, in which case I think it might be too much for the arduino as the keys are already 1/8 of the ram.

Let me know if you want to pursue it and I could take a closer look.

NicolasLiveris, 2020-04-24:
@gguuss and @cirvladimir 
Thank you for your response. However, I gave up because I didn't have enough time and lacked knowledge at the time. I opted for a compatible card.
Thank you for taking the time to answer.

closed 2020-04-24 by gguuss