GoogleCloudPlatform/google-cloud-iot-arduino #156: coming up with this error but then will sometimes randomly connect to library

nzoffgrid, 2020-04-21:
coming up with this error but then will sometimes randomly connect to library

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
Connecting to WiFi
Waiting on time sync...

connecting...Refreshing JWT

User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: bearssl
sp: 3fff21a4 end: 3fff2224 offset: 0000
3fff21a4:  40102692 000000d5 00000004 deadbeef  
3fff21b4:  000000fe 00000000 00000000 00000000  
3fff21c4:  00000000 00000000 00000000 3fff21f4  
3fff21d4:  00000004 00000090 00000004 40102692  
3fff21e4:  00003a98 00000000 3fff0a4c 4020a172  
3fff21f4:  3ffe92ea 00000000 40222694 4020a188  
3fff2204:  3fff8299 3fff3bdc deadbeef 40209321  
3fff2214:  deadbeef 00000000 3fff0a4c 40204ee8  

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 0000
3ffffdd0:  00003a98 00000000 00000008 40207df2  
3ffffde0:  000005f3 00000005 cee9377f 00000000  
3ffffdf0:  00000001 000011a6 3fff33ec 4022206e  
3ffffe00:  3fff38ec 3fff22c4 3fff33ec 00000001  
3ffffe10:  00000001 00000001 3fff0a4c 40208030  
3ffffe20:  3fff22c4 00000000 3fff0a4c 40208213  
3ffffe30:  000000b4 3ffefcc4 00000001 40209c67  
3ffffe40:  00000000 00000000 00000000 4020a26a  
3ffffe50:  00000000 3fff24f4 3fff0a4c 40206e08  
3ffffe60:  000022b3 00000d50 3fffff00 3fff2d84  
3ffffe70:  000022b3 3fff0a4c 3fff22c4 3fff2d84  
3ffffe80:  000022b3 3fff0a4c 3fff22c4 40208331  
3ffffe90:  4020c218 cebce940 4020c218 cebce940  
3ffffea0:  3fff2e1c 3ffe9066 00000000 402088cc  
3ffffeb0:  00000000 000000c6 3fffff30 402096d8  
3ffffec0:  3fff3164 3ffef0fc 3fffff30 40209850  
3ffffed0:  3ffefb04 3ffef0fc 3fffff30 40209880  
3ffffee0:  3ffef0fc 3ffef0fc 3fffff30 4020951c  
3ffffef0:  3ffefb04 3ffef108 3fffff30 40202272  
3fffff00:  3fff2b4c 00000000 80fe002f 6a6f7200  
3fffff10:  73746365 80fe002f 3fff228c 3fff2d84  
3fffff20:  00000000 3ffefb24 3fff228c 40202de4  
3fffff30:  3fff2e1c 00c600cf 00ff0a00 3fff2b4c  
3fffff40:  0050005f 00044243 3ffefcc4 00000000  
3fffff50:  3fff2b4c 3ffefcc4 00000014 3ffefc64  
3fffff60:  402419ce 042c1d80 3fff0a4c 3ffefc64  
3fffff70:  3fffdad0 00000000 3ffefb04 402023b6  
3fffff80:  0001c200 00000000 3ffefc24 402024d6  
3fffff90:  3fffdad0 00000000 3ffefc24 40202492  
3fffffa0:  feefeffe feefeffe 3ffefc24 40209d7c  
3fffffb0:  feefeffe feefeffe 3ffe8514 40100c09  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
Connecting to WiFi
Waiting on time sync...

connecting...Refreshing JWT
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/iotgumboot/locations/europe-west1/registries/iot-registry/devices/esp82
Waiting 60 seconds, retry will likely fail

DanielGarc, 2020-04-21:
You probably need to extend the BSLL stack:
https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/138#issuecomment-562391042

raphael-bmec-co, 2020-05-27:
This can occur when you run out of stack. You should check how much stack you have as you move through the connection process.

gguuss, 2020-10-15:
Closing as this issue is stale and the resolution documented for **StackThumk.cpp** in the Github README should resolve this.

closed 2020-10-15 by gguuss