GoogleCloudPlatform/google-cloud-iot-arduino #157: ESP8266: Settings incorrect or missing a cyper for SSL

mfx22, 2020-04-30:
I am creating an IoT device that sends data to GCP running on ESP8266 (Sparkfun ESP8266 Thing).

I'm using the Arduino IDE and SPIFFS to upload the ca file and also making use of the EEPROM in conjunction with SmartConfiig to store the login credentials for wifi. 

When the code to read EEPROM is not in the code, the device can transmit to GCP no problem.  However, when I add that code to read credentials off of EEPROM in, I am getting this error:

```
connecting...Refreshing JWT
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
```

My initial reaction is that somehow the EEPROM read is affecting the CA file.  But this doesn't make sense since filesystem and EEPROM are separate in memory.  Any ideas to what could be causing this issue?

tyczj, 2020-05-06:
I get this same issue on ESP32, Sometimes it will reconnect just fine but other times it just keeps trying to reconnect infinitely. It usually takes a few days for it to fail like this though. It seems like something ends up being wrong with the JWT

galz10, 2020-05-13:
So a couple things you could do to check whats going wrong. The first is printing out the JWT to see if it has the right information in it (expiration ,correct project id) , you can check this by going to jwt.io.

Another thing you can do is make sure your device has the correct private key and that the device on GCP IoT Core has the matching public key. Lastly depending on your board your wifi connectivity might be a bit finicky so personally sometimes standing close to the router helped.

raphael-bmec-co, 2020-05-27:
This can occur when you run out of stack. You should check how much stack you have as you move through the connection process.

FedeBev, 2020-07-09:
Same problem here with arduino mkr 1010

@raphael-bmec-co could you please provide some more information? Thanks

raphael-bmec-co, 2020-07-09:
Hi @FedeBev.

Please see #144. Let me know if you don't come right based on that information.

FedeBev, 2020-08-08:
This issue is really hard to debug for me.

Following the hit that @raphael-bmec-co  gave me, I reduced the stack memory pressure by moving some instances (unrelated to this library) in the heap and the situation seems to improve because now the connection fails only the first time and the reconnection when the network fails for any reason also works.

That's good, the problem is that my stack was really small and there was only a few very slight class instances. That being said, can the problem be in the function that handles the obtaining of the token?

GulzaibKhan-BSEE17021, 2021-05-19:
Hey i am facing the same problem, can any of you tell that how was stack size increased?

FedeBev, 2021-05-19:
The end of nightmare for me, I hope this will help someone.

Using the following code on ESP8266 after a failed connection I was able to immediately identify the issue, simply a silly wrong epoch in my case

```c++
char buf[256];
auto code = netClient.getLastSSLError(buf, 256);
if (code != 0)
{
    Serial.print("Last SSL ERROR was: ");
    Serial.print(code);
    Serial.print(" - ");
    Serial.println(buf);
}
```

ChaserVasya, 2021-11-13:
Solution
===

In my case it was a little timeout. In example there is:
```
    mqttClient->setOptions(180, false, 1000); // keepAlive, cleanSession, timeout
```
I change it to 10000 and it works! 
BUT first connection always fails. I don`t know why.

Details
===
In my case `mqttClient->lastError()` returns 
LWMQTT_NETWORK_FAILED_WRITE 
LWMQTT_MISSING_OR_WRONG_PACKET

LWMQTT_MISSING_OR_WRONG_PACKET on connecting is setted by 
`MQTTClient::connect`  -> ` lwmqtt_connect` -> `lwmqtt_cycle_until`.
`lwmqtt_cycle_until` processes input during timeout and if nothing have been inputted it returns 
`LWMQTT_SUCCESS` and  `LWMQTT_NO_PACKET `
`lwmqtt_connect` when there are those values returns `LWMQTT_MISSING_OR_WRONG_PACKET`.
Finally, `MQTTClient::connect` returns  `Settings incorrect or missing a cyper for SSL`. 

Propose
===
I HAVE SPEND ~50 HOURS ON THIS PROBLEM. I HAVE LEARNED ALL ABOUT SSL, TLS, CERTIFICATES, RE-READ LIB CODE AND CODE OF SUBLIBS MANY TIMES. TRY ALL CRUTCHES.

I could have avoided all this if the error logs were a little more specific. In my case, nothing related to certificates, cyper. 
Yes, `timeout` is a settings. But there are thousands other settings. ESPECIALLY in netClient`s crypro. 

Collaborators, could you do more informative logs?