GoogleCloudPlatform/google-cloud-iot-arduino #159: ESP32 stuck on "checking wifi.."

RyanGaudion, 2020-05-06:
I've followed the following tutorial in order to connect my esp32 to Iot Core:
https://www.survivingwithandroid.com/cloud-iot-core-esp32/

When opening serial monitor I get constant "checking wifi..." messages. What could be the issue? Why trying other Wifi related scripts on the ESP32, they connect without issue.

Please let me know if you need anymore information:

Arduino IDE - 1.8.12

galz10, 2020-05-24:
What version of the Google Cloud IoT library are you using ? 
The Google Cloud IoT Core JWT 1.1.9 version connects to the WiFi and to the cloud, if this is the current version you have i would check what WiFi you're trying to connect since the ESP32 is better on 2.4Ghz connection.

RyanGaudion, 2020-05-25:
Thank you, pulling the latest version worked. Thank you for the fix

closed 2020-05-25 by RyanGaudion