GoogleCloudPlatform/google-cloud-iot-arduino #161: Support for port 443 on LTS

raphael-bmec-co, 2020-05-12:
Google now supports port 443 on LTS but it is hardcoded to 8883. It would be great to make this configurable. 

Thanks.

raphael-bmec-co, 2020-05-29:
It is possible to work around this by calling the following directly instead of using `mqtt_->startMQTT()`:
` mqtt_client_->begin(<HOST>, <PORT>, *net_client_);`

If you do so, you need to set the callback using:
`mqtt_client_->onMessage(on_message);` if you are happy with a global handler.
`mqtt_client_->onMessageAdvanced(on_message);` if you need a reference to the client to recover the calling instance.

felixback84, 2021-03-10:
Hello guys I have an ESP32 LilyGo T-Call SIM800 Series module with GSM and I want to send messages from it to GCP's IoT Core, for this I have used two TinyGSM libraries and
Google_Cloud_IoT_Core_JWT. Both do have been useful separately, that is, with TinyGSM I have managed to send messages via GSM to a server (using the examples it brings) and with Google_Cloud_IoT_Core_JWT I have managed to send MQTT messages via WiFi to GCP's IoT Core. However, the problem is when I try to do both at the same time. Since I always have the same error (the return returned is: Settings incorrect or missing a cyper for SSL) I have read this post and the one associated with it carefully and I understand that it is normal to use the Google_Cloud_IoT_Core_JWT library since it cannot establish a secure connection between the device and GCP, as I have seen and with I have investigated has something to do with the secure connection using an SSL certificate. Reading the comments of @Wiz-IO I have tried to intuit things, but nevertheless I have not been able to make it work, since although I know that he has succeeded, I cannot fully spin his suggestions on how to solve it, please do know What should I do? I ask you to help me, since it is important for a project of my company. For me everything happens in this part of the code. 

I take this code extract from the related post since I have tried to do almost exactly the same thing, with the same results

```
Client* client = new TinyGsmClient(modem);
setupCloudIoT(client);

void setupCloudIoT(Client * _tinyGSMClient) {
    device = new CloudIoTCoreDevice(
    project_id, location, registry_id, device_id,
    private_key_str);
         
    netClient =  _tinyGSMClient;
    mqttClient = new MQTTClient(512);
    mqttClient->setOptions(180, true, 1000); // keepAlive, cleanSession, timeout
    mqtt = new CloudIoTCoreMqtt(mqttClient, netClient, device);
    mqtt->setUseLts(true);
    mqtt->startMQTT();
}
```
------------------------------
**the return returned is: Settings incorrect or missing a cyper for SSL**

Thank you in advance in case any of those present in these post have managed to solve it