GoogleCloudPlatform/google-cloud-iot-arduino #165: How to push to a specific topic?

RyanGaudion, 2020-05-25:
I'm currently using the following line which is working well:
`publishTelemetry(getDefaultSensor());`

However, I see the following line commented in the example:
`publishTelemetry(mqttClient, "/sensors", getDefaultSensor());`
However, when I uncomment it I get the following error:

> Test2:46:68: error: no matching function for call to 'publishTelemetry(MQTTClient*&, const char [12], String)'
> 
>      publishTelemetry(mqttClient, "/data-topic", getDHTsensorValue());
> 
>                                                                     ^
> 
> 
> sketch\esp32-mqtt.h:88:6: note: candidate: bool publishTelemetry(String)
> 
>  bool publishTelemetry(String data) {
> 
>       ^
> 
> sketch\esp32-mqtt.h:88:6: note:   candidate expects 1 argument, 3 provided
> 
> sketch\esp32-mqtt.h:92:6: note: candidate: bool publishTelemetry(const char*, int)
> 
>  bool publishTelemetry(const char* data, int length) {
> 
>       ^
> 
> sketch\esp32-mqtt.h:92:6: note:   candidate expects 2 arguments, 3 provided
> 
> sketch\esp32-mqtt.h:96:6: note: candidate: bool publishTelemetry(String, String)
> 
>  bool publishTelemetry(String subfolder, String data) {
> 
>       ^
> 
> sketch\esp32-mqtt.h:96:6: note:   candidate expects 2 arguments, 3 provided
> 
> sketch\esp32-mqtt.h:100:6: note: candidate: bool publishTelemetry(String, const char*, int)
> 
>  bool publishTelemetry(String subfolder, const char* data, int length) {
> 
>       ^
> 
> sketch\esp32-mqtt.h:100:6: note:   no known conversion for argument 1 from 'MQTTClient*' to 'String'
> 
> Multiple libraries were found for "WiFi.h"
>  Used: C:\Users\x\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
>  Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
> exit status 1
> no matching function for call to 'publishTelemetry(MQTTClient*&, const char [12], String)'

galz10, 2020-05-26:
If you want to publish data to a specific spot like "/sensors" you will need to make another topic in PubSub and associate this topic with a subfolder in your registry, for example, the topic name could be "temperature" and the subfolder name can be "sensors" now if you use the function below you should see the telemetry data only in the subscription that is associated to the subfolder

`bool publishTelemetry(String subfolder, String data) {
  return mqtt->publishTelemetry(subfolder, data);
}`

you can call this function by using  `publishTelemetry("/sensor",getDefaultSensor());`,
make sure that when you're setting up the new topic that you also have a subscription in it. 

I also attached a link that explains what's the purpose of subfolders if you have questions about that. 

Link: https://cloud.google.com/iot/docs/how-tos/devices

gguuss, 2020-10-15:
This looks stale, closing.

closed 2020-10-15 by gguuss