GoogleCloudPlatform/google-cloud-iot-arduino #167: Replace need for globally defined functions and variables

raphael-bmec-co, 2020-05-27:
Globally defined functions and variables should ideally be passed in the constructor.

raphael-bmec-co, 2020-05-31:
This is the loop function:

```
void CloudIoTCoreMqtt::loop() {
  if (millis() > device->getExpMillis() && mqttClient->connected()) {
    // reconnect
    Serial.println("Reconnecting before JWT expiration");
    iat = 0; // Force JWT regeneration
    getJwt(); // Regenerate JWT using device function
    mqttClient->disconnect();
    mqttConnect(true); // TODO: should we skip closing connection
  }
  this->mqttClient->loop();
}
```

`mqttClient->disconnect();` is called which stops the net client. But then `mqttConnect(true);` is called with the skip parameter `true`. Which means the net client is not reconnected. I may well be missing something, but it would make sense that it has to wait till the next loop where the parameter is set `false` and the reconnect is made. 

Haven't tested any of this. Just something I came across in the rewrite.

galz10, 2020-06-12:
Hi Raphael, 
Thank you for the feedback, correct me if i'm wrong but what you're asking is if you could set the callback inside of mqttClient correct ? 

Also for onMessageAdvanced i'm looking at the code on mqttClient.cpp and it doesn't return a pointer, it sets the advanced callback to whatever function you set. 

I haven't tried setting the callback in mqttClient.h but theoretically it should work, could you maybe give an example function so i could test to see the issue you're referencing?

raphael-bmec-co, 2020-06-12:
Hi Gal

Thanks for getting back to me.

Since posting, I have worked around both of these issues by not using the `CloudIoTCoreMqtt` layer all together. So I'm all set on my side but I've explained in more detail below in case you want to make any changes to the `CloudIoTCoreMqtt` layer.

Ultimately, what I am suggesting is that the `CloudIoTCoreMqtt` layer follows the same pattern as the `MQTTClient`. 

The `CloudIoTCoreMqtt` object could be instantiated something like:

```
// Example message callback declaration.
static void on_message_received(CloudIoTCoreMqtt *cloud_iot_core, String message){
// Handle your message.
}

// Example get JWT callback declaration.
static String get_jwt(CloudIoTCoreMqtt *cloud_iot_core){
// Return generated JWT.
}

// Example constructor (could be replaced with setters).
auto* cloud_iot_core_mqtt = CloudIoTCoreMqtt(on_message_received,  get_jwt);
```

When I referred to returning a pointer, I am talking about the `CloudIoTCoreMqtt` pointer in the `on_message_received` and `get_jwt` example callbacks above.

Only pointers to **static** methods can be passed as parameters. This means that the callback - for instance `on_message_received` - cannot directly reference the instance of the class which defined the static callback. Specifically we cannot access `this` or any of its members inside the callback.
```
static void on_message_received(CloudIoTCoreMqtt *cloud_iot_core, String message){
  // We cannot access "this" here.
}
```

There is a standard pattern to work around this which is implemented in the `MQTTClient` layer.

The method would required the `CloudIoTCoreMqtt` class to contain a variable `void*ref` which is public and unused by the `CloudIoTCoreMqtt` class. This allows the instantiating layer to set this `ref` equal to a pointer to said instance of said instantiating class i.e. `this` through the pointer to the `CloudIoTCoreMqtt`:

 ```
static void on_message_received(CloudIoTCoreMqtt *cloud_iot_core, String message){
  IntantiatingClass *p_class = (IntantiatingClass *) cloud_iot_core->ref;
  p_class.doSomeThingWithMessage(message);
}
```

Maybe I'm missing something and there is a reason for the use of global functions instead of passed callbacks but that's my two cents. Keen to hear your thoughts.

On a more urgent note, please see my comments in #144. There is a major bug with the `CloudIotCoreDevice.createJWT()` and the use of strings there. Looking forward to a fix there!

galz10, 2020-06-12:
okay thanks for your feedback i'll take a look at #144

sonyccd, 2020-11-25:
I would also like to see a passed callbacks.