GoogleCloudPlatform/google-cloud-iot-arduino #170: Unstable connection using esp8266 and this library.

xiskoct, 2020-05-30:
When connecting esp8266 with google iot core I get in the gcp logging several connection attempts. Sometimes it connects and publishes, but sometimes it makes a sequence of connections without publishing. The error I receive is:

{"code":9,"description":"FAILED_PRECONDITION"},"protocol":"MQTT

[image]

raphael-bmec-co, 2020-05-30:
Can you set the debugging level on the ESP8266 to verbose and share the corresponding device logs.

xiskoct, 2020-05-30:
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with pelon, channel 1
dhcp client start...
ip:192.168.1.159,mask:255.255.255.0,gw:192.168.1.1
Waiting on time sync...
1667592820
Disconnect mqtt

connecting...Refreshing JWT
:ref 1
:wr 227 0
:wrc 227 227 0
:ack 227
:rn 536
:rch 536, 536
:rd 5, 1072, 0
:rdi 536, 5
:rch 1072, 536
:rch 1608, 174
:rd 87, 1782, 5
:rdi 531, 87
:rd 5, 1782, 92
:rdi 444, 5
:rd 1523, 1782, 97
:rdi 439, 439
:c 439, 536, 1782
:rdi 536, 536
:c 536, 536, 1246
:rdi 536, 536
:c 536, 536, 710
:rdi 174, 12
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/stunning-symbol-272402/locations/us-central1/registries/atest-registry/devices/atest-dev
Waiting 60 seconds, retry will likely fail
pm open,type:2 0
:rcl pb=0x3fffa034 sz=174
Refreshing JWT
:close
:ur 1
:dsrcv 174
:del
:ref 1
:wr 227 0
:wrc 227 227 0
:ack 227
:rn 536
:rch 536, 536
:rd 5, 1072, 0
:rdi 536, 5
:rch 1072, 536
:rch 1608, 174
:rd 87, 1782, 5
:rdi 531, 87
:rd 5, 1782, 92
:rdi 444, 5
:rd 1523, 1782, 97
:rdi 439, 439
:c 439, 536, 1782
:rdi 536, 536
:c 536, 536, 1246
:rdi 536, 536
:c 536, 536, 710
:rdi 174, 12
:rd 5, 174, 12
:rdi 162, 5
:rd 114, 174, 17
:rdi 157, 114
:rd 5, 174, 131
:rdi 43, 5
:rd 29, 174, 136
:rdi 38, 29
:rd 5, 174, 165
:rdi 9, 5
:rd 4, 174, 170
:rdi 4, 4
:c0 4, 174
:wr 49 0
:wrc 49 49 0
:wr 6 0
:wrc 6 6 0
:wr 45 0
:wrc 45 45 0
:ack 49
:ack 51
:rn 51
:rd 5, 51, 0
:rdi 51, 5
:rd 1, 51, 5
:rdi 46, 1
:rd 5, 51, 6
:rdi 45, 5
:rd 40, 51, 11
:rdi 40, 40
:c0 40, 51
:wr 365 0
:wrc 365 365 0
:ack 365
:close
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/stunning-symbol-272402/locations/us-central1/registries/atest-registry/devices/atest-dev
Waiting 60 seconds, retry will likely fail
Refreshing JWT
:ur 1
:dsrcv 0
:del
:ref 1
:wr 227 0
:wrc 227 227 0
:ack 227
:rn 536
:rch 536, 536
:rch 1072, 536
:rch 1608, 174
:rd 5, 1782, 0
:rdi 536, 5
:rd 87, 1782, 5
:rdi 531, 87
:rd 5, 1782, 92
:rdi 444, 5
:rd 1523, 1782, 97
:rdi 439, 439
:c 439, 536, 1782
:rdi 536, 536
:c 536, 536, 1246
:rdi 536, 536
:c 536, 536, 710
:rdi 174, 12
:rd 5, 174, 12
:rdi 162, 5
:rd 114, 174, 17
:rdi 157, 114
:rd 5, 174, 131
:rdi 43, 5
:rd 29, 174, 136
:rdi 38, 29
:rd 5, 174, 165
:rdi 9, 5
:rd 4, 174, 170
:rdi 4, 4
:c0 4, 174
:wr 49 0
:wrc 49 49 0
:wr 6 0
:wrc 6 6 0
:wr 45 0
:wrc 45 45 0
:ack 49
:ack 51
:rn 51
:rd 5, 51, 0
:rdi 51, 5
:rd 1, 51, 5
:rdi 46, 1
:rd 5, 51, 6
:rdi 45, 5
:rd 40, 51, 11
:rdi 40, 40
:c0 40, 51
:wr 365 0
:wrc 365 365 0
:ack 365
:close
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/stunning-symbol-272402/locations/us-central1/registries/atest-registry/devices/atest-dev
Waiting 60 seconds, retry will likely fail
Refreshing JWT
:ur 1
:dsrcv 0
:del
:ref 1
:wr 227 0
:wrc 227 227 0
:ack 227
:rn 536
:rch 536, 536
:rch 1072, 536
:rch 1608, 174
:rd 5, 1782, 0
:rdi 536, 5
:rd 87, 1782, 5
:rdi 531, 87
:rd 5, 1782, 92
:rdi 444, 5
:rd 1523, 1782, 97
:rdi 439, 439
:c 439, 536, 1782
:rdi 536, 536
:c 536, 536, 1246
:rdi 536, 536
:c 536, 536, 710
:rdi 174, 12
:rd 5, 174, 12
:rdi 162, 5
:rd 114, 174, 17
:rdi 157, 114
:rd 5, 174, 131
:rdi 43, 5
:rd 29, 174, 136
:rdi 38, 29
:rd 5, 174, 165
:rdi 9, 5
:rd 4, 174, 170
:rdi 4, 4
:c0 4, 174
:wr 49 0
:wrc 49 49 0
:wr 6 0
:wrc 6 6 0
:wr 45 0
:wrc 45 45 0
:ack 49
:ack 51
:rn 51
:rd 5, 51, 0
:rdi 51, 5
:rd 1, 51, 5
:rdi 46, 1
:rd 5, 51, 6
:rdi 45, 5
:rd 40, 51, 11
:rdi 40, 40
:c0 40, 51
:wr 365 0
:wrc 365 365 0
:ack 365
:rn 33
:rd 5, 33, 0
:rdi 33, 5
:rd 28, 33, 5
:rdi 28, 28
:c0 28, 33
connected

Library connected!
:wr 61 0
:wrc 61 61 0
:ack 61
:rn 34
:rd 5, 34, 0
:rdi 34, 5
:rd 29, 34, 5
:rdi 29, 29
:c0 29, 34
:wr 65 0
:wrc 65 65 0
:ack 65
:rn 34
:rd 5, 34, 0
:rdi 34, 5
:rd 29, 34, 5
:rdi 29, 29
:c0 29, 34
:wr 66 0
:wrc 66 66 0
:ack 66
:wr 84 0
:wrc 84 84 0
:ack 84
Publicando telemetria
:wr 69 0
:wrc 69 69 0
:ack 69
:rn 63
:rch 63, 63
:rd 5, 126, 0
:rdi 63, 5
:rd 58, 126, 5
:rdi 58, 58
:c 58, 63, 126
incoming: /devices/atest-dev/config - led
Switch led off
:rd 5, 63, 0
:rdi 63, 5
:rd 58, 63, 5
:rdi 58, 58
:c0 58, 63
:wr 33 0
:wrc 33 33 0
:ack 33
incoming: /devices/atest-dev/config - led
Switch led off
:wr 33 0
:wrc 33 33 0
:ack 33
Publicando telemetria
:wr 69 0
:wrc 69 69 0
:ack 69
17:44:05.225 -> Publicando telemetria
17:44:05.225 -> :wr 69 0
17:44:05.225 -> :wrc 69 69 0
17:44:05.258 -> :ack 69

raphael-bmec-co, 2020-05-30:
Nothing obvious is jumping out at me. Please move over to #144 as it seems like the same/similar issue and there are several people working on it.

galz10, 2020-06-23:
Hi @xiskoct have you managed to fix this issue ?

meltdown03, 2020-08-11:
For my esp8266, I had to edit the `_stackSize` from `(6200/4)` to `(6536/4)` in the `StackThunk.cpp` file:
https://github.com/esp8266/Arduino/blob/777323709b5e49d22ee611e6ec10b2ddc69a5a66/cores/esp8266/StackThunk.cpp#L43

gguuss, 2020-10-15:
There have been enhancements / improvements (base64 fix, advanced callback, fix for when messages are empty in esp8266) that may have resolved this. I believe the StackThunk.cpp fix is also provided in the README.

closed 2020-10-15 by gguuss