GoogleCloudPlatform/google-cloud-iot-arduino #172: duplication of state messages

biancoblu79, 2020-06-13:
Hello, 

I am getting duplicated device "state" messages and I believe that's because of of sending them to both the state and the telemetry topics upon connection ... 

void CloudIoTCoreMqtt::onConnect() {
  if (logConnect) {
    publishState("connected");
    publishTelemetry("/events", device->getDeviceId() + String("-connected"));
  }
}

is there a specific reason for sending state messages to the telemetry topic? I commented out the publishTelemetry line for my use case.. but was wondering if there was a particular reason for having that line of code.

thanks! 
marco

galz10, 2020-06-16:
Hi Marco thanks for the feedback, 
there isn't a specific reason for sending state message and telemetry message other then just having the ability to send the two messages, but the code above shouldn't send a duplicated state message unless you're calling onConnect() multiple times.

biancoblu79, 2020-06-18:
Hi galz10, thanks for the reply. I actually need to send these messages and to 2 different subfolders in Cloud IoT Core. I was just wondering why the state message was sent, together with device data, to the default telemetry folder. I just commented a line of code out and I am sending the state message separately from the telemetry data.. thanks!

closed 2020-06-18 by biancoblu79