GoogleCloudPlatform/google-cloud-iot-arduino #173: Can't publish to mqtt

TimAtFacture, 2020-06-25:
Hello all,

I am having trouble getting my program to publish data over mqtt. I have had this working in the past, but it seems to only work on certain WiFi networks, and it now doesn't work on WiFi networks where it has previously. I have traced the error to the publishTelemetry() function returning '0'. I have started poking around in the arduino-mqtt library to see what is going on, but I can't find the mqtt->publishTelemetry() function declaration which the publishTelemetry() function calls.

I sometimes see error messages come from the serial, but they don't seem to be consistent. I sometimes see errno 118 and errno 5. I haven't been able to find a place where these error messages are described yet.

galz10, 2020-06-26:
Hi,
I know that the ESP32 and EPS8266 have problems on 5Ghz wifi, also could you provide a picture of the error?

gguuss, 2020-10-15:
Issue seems stale, please re-open and provide further details such as the board you are using and versions of the library and hardware.

closed 2020-10-15 by gguuss