GoogleCloudPlatform/google-cloud-iot-arduino #177: Token  do not created  correctly.

aleksandr-evgenevich, 2020-07-15:
Token  do not created  correctly. 
I use the part of the library that creates tokens and check the correctness on the site jwt.io
i use arduino due.  I am not sure that the choice of the board matters because I do not use mqtt.

Body:
{
  "alg": "ES256",
  "typ": "JWT"
}
{
  "iat": 1594738479,
  "exp": 1594742079,
  "aud": "1234"
}

Correct from jwt.io:
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1OTQ3Mzg0NzksImV4cCI6MTU5NDc0MjA3OSwiYXVkIjoiMTIzNCJ9.nVi6ySobQ5FsYCKt47tTBTZjcwrQ5xx32c-fP1qzdratWXpDuIji6YG5cVCk8Plv-sFNo-5c1R90TK45GW5nIA

Incorrect (created using the library):
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1OTQ3Mzg0NzksImV4cCI6MTU5NDc0MjA3OSwiYXVkIjoiMTIzNCJ9.rUz5/s55ijSNm18pMpPXHAuIVKsyP+HsIH4go8I0Q7BCvrje4FC03hfwo7PlFWnI4oI19cb470dvS8Xui9texg==

read EC key
Private-Key: (256 bit)
priv:
    eb:a0:bf:19:6d:88:d8:9f:b9:b6:8d:f5:fe:1c:b0:
    f4:ac:6b:da:4b:8a:b4:04:b1:da:de:3d:26:a2:71:
    3c:77
pub:
    04:dd:31:7e:d8:88:ec:4e:dd:bf:e9:5d:d9:dc:bf:
    2a:34:35:f2:a5:5f:05:2f:0e:97:47:a3:11:7f:b6:
    1a:f1:16:45:85:24:c6:c5:89:73:9d:7b:18:40:63:
    a5:2c:9c:56:98:09:6f:2d:19:a9:2c:34:de:38:6e:
    be:39:f8:0c:ec
ASN1 OID: prime256v1
NIST CURVE: P-256

ronyeh3, 2020-07-15:
maybe related that to fact that the SHA function is this project is made on base64 , and its should  be actuality should be made on base64URL

raphael-bmec-co, 2020-07-15:
Have a look at #144

galz10, 2020-07-19:
Hi @aleksandr-evgenevich when you tried using the JWT that was created from the code you used from the library did Cloud IoT Core reject the connection ? Did you receive a error ?

ronyeh3, 2020-07-19:
Hi Gal, greeting from Israel
as @aleksandr-evgenevich mention, we didn't use this library for MQTT , therefore no connection to google made.
we use it just to sign our requests with es256.
we manage to overcome our problems, 2 things we saw:

1. #144 , SHA256 is completely corrupted due to its null terminated string, so we converted it to bytes.

2. we use base64url and not base64, that is everybody else standard.

then, all working good.

`    // Making jwt token json
    String header = "{\"alg\":\"ES256\",\"typ\":\"JWT\"}";
    String payload = "{\"iat\":" + int_to_string(time) +
                     //  ",\"exp\":" + int_to_string(time + exp_secs) +
                     ",\"sn\":\"" + project_id + "\"}";

    String header_payload_base64 =
        base64_encode(header) + "." + base64_encode(payload);

    trace(F("jwt: payload: %s"), header_payload_base64.c_str());

    Sha256 sha256Instance;
    sha256Instance.update((const unsigned char *)header_payload_base64.c_str(), header_payload_base64.length());
    unsigned char sha256[SHA256_DIGEST_LENGTH];
    sha256Instance.final(sha256);
    trace(F("jwt: sha256: %s"), string_to_hex((char *)sha256, sizeof(sha256)).c_str());

    ecc_init();

    // Signing sha with ec key. Bellow is the ec private key.
    point_t pub_key;
    ecc_gen_pub_key(priv_key, &pub_key);

    ecdsa_init(&pub_key);

    NN_DIGIT signature_r[NUMWORDS], signature_s[NUMWORDS];
    ecdsa_sign((uint8_t *)sha256, signature_r, signature_s, priv_key);

    auto rc = ecdsa_verify((uint8_t *)sha256, signature_r, signature_s, &pub_key);
    trace(F("jwt: ecdsa verify: %d"), rc);

    String result = header_payload_base64 + "." +
                    MakeBase64Signature(signature_r, signature_s);
    trace(F("jwt: token: %s"), result.c_str());
`

galz10, 2020-07-19:
Hey @ronyeh3 , looking at your code it doesn't seem different from what we have on our jwt.cpp

galz10, 2020-07-24:
Issue was fixed by Alvaro, so i'm closing this issue.

closed 2020-07-24 by galz10