GoogleCloudPlatform/google-cloud-iot-arduino #183: ESP8266 Compilation Error

coolcoastcat, 2020-08-07:
I'm using the example code and I believe I'm using the correct MQTT library (platformio id #617).  However, when I attempt to build I'm receiving the following compilation error:
```
/home/pi/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/esp12e/libc5e/libGoogle Cloud IoT Core JWT.a(CloudIoTCoreMqtt.cpp.o):(.text._ZN16CloudIoTCoreMqtt9startMQTTEv+0xc): undefined reference to `messageReceived(String&, String&)'
```

I've narrowed it down to the following line of code from the example code provided: 
`mqtt->startMQTT(); // Opens connection`

After poking into the CloudIoTCoreMqtt.cpp and MQTT library, I'm not sure why this is occurring. My c++ skills have some heavy rust on them so if I'm overlooking something obvious, I'd appreciate a looksee.  I have  the following lib_deps defined in my platformio.ini file:

 ```
 MQTT@2.4.7
  Google Cloud IoT Core JWT@1.1.10
```

FedeBev, 2020-08-08:
Maybe the problem is that you don't have defined the function `messageReceived(String&, String&)`

The implementation of the function `messageReceived(String&, String&)` is up to you, there's just a declaration in `CloudIoTCoreMqtt.cpp` but not a definition.

Did you see the example folder? This, for instance, should help: https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/MKR1000-lwmqtt/mkr1000-mqtt.h

gguuss, 2020-10-02:
I'm going to close this, the example code compiles for me and CI still seems to be working. Thank you @FedeBev for the extra help.

closed 2020-10-02 by gguuss