GoogleCloudPlatform/google-cloud-iot-arduino #184: Create topic from ESP8266

ajlop, 2020-08-08:
Hi! I want to use the MAC address of the devices to create individuals topic of each one.
I am migrating my mqtt broker to google cloud and this is how I had it implemented (each new device created its own topic using its MAC address). can i use that architecture in google cloud? How can I implement it in GCP?

Thanks for your help !!

gguuss, 2020-10-06:
You would need to use a service account to automatically add the device to the registry that you will be using for your data. This can be done using Cloud Functions, DataFlow, or other cloud compute-engine focused solutions. This is not a core part of IoT at present.

gguuss, 2020-10-15:
Closing as this is out of scope for this project but marking as enhancement to inform the product team.

closed 2020-10-15 by gguuss