GoogleCloudPlatform/google-cloud-iot-arduino #186: How to subscribe to a new topic via Code

mars000, 2020-08-09:
Hello

I'm wish to subscribe to a specific topic.  Can some provide an example snippet that works ?
Also - can I create a new topic using a function call in from my ESP32 ?

I'm using something like this but always get a FALSE returned and therefore SUBSCRIBE: fail message.

My code snippet:

```
if (mqttClient->subscribe("/devices/led-dev/events",1))
   {
    Serial.println("[1] SUBSCRIBE: ok");
  } else
  {
    Serial.println("[1] SUBSCRIBE: fail");
  }
```

these are my project details:
```
getEventsTopcs() = /devices/led-dev/events
getCommandsTopic() = /devices/led-dev/commands/#
getDeviceId() = led-dev

```

gguuss, 2020-08-10:
You can only subscribe to the special topics `/devices/{device-id}/config` and `/devices/{device-id}/commands`.

You can only publish to the special topics `/devices/{device-id}/state` and `/devices/{device-id}/events`.

The MQTT interface is not designed for creating read/write queues shared between devices, this must be done with some logic hosted on Google Cloud.

You can learn more about this in [the devices conceptual documentation](https://cloud.google.com/iot/docs/concepts/devices).

mars000, 2020-08-10:
Thanks Gus - that’s disappointing :-(

So I can’t subscribe to a sub folder or publish to a sub folder ?
Are there any real examples of how to do this with google IoT and esp32 ?

gguuss, 2020-08-12:
Yup, one of the trade-offs that must be made in order to operate efficiently at the scale of IoT Core.

There are a ton of Community articles that cover advanced scenarios, you might want to check out [Device-to-device communication](https://cloud.google.com/community/tutorials/iot-device-to-device).

mars000, 2020-08-12:
Thanks Gus – much appreciated.

Apologies to labour on this as some confusion for me.

I’ve actually created new topics and subscriptions on Google IoT and Registry. See screenshot below:

Just to be clear I’m trying to publish to these subfolders using your library but with no success.

Also (more importantly) I’m trying to subscribe to messages from specific Topics e.g. /sensor but its not working.

Can you kindly provide example of how I would so this ?

 

I’m using below But its not working…(am I doing it correctly ?)

  publishTelemetry("/temp", "43210");

 

For subscription my statement is below but it fails….

this->mqttClient->subscribe("/devices/led-dev/sensor", 0)

gguuss, 2020-10-06:
# Subscribe
* You can only subscribe to the configuration and command topic for your device.

# Publish
* You can only publish to the events and state topics for your device.

Trying to go outside of the protocols and publish to non-device topics will cause the bridge to disconnect you. More information can be found at https://cloud.google.com/iot/docs/how-tos/mqtt-bridge

closed 2020-10-06 by gguuss