GoogleCloudPlatform/google-cloud-iot-arduino #187: EC certificates don't work anymore?

datascraps, 2020-08-16:
I get this error following the guides the best I can
`ERROR: (gcloud.iot.devices.create) INVALID_ARGUMENT: The device credential in position 0 must be an X.509 certificate.`

I think this library only supports EC keys and clout iot only supports RSA with at least 2048 bits and NIST P-256?
https://cloud.google.com/iot/docs/how-tos/credentials/verifying-credentials

This is the script I'm running to generate everything 
```
openssl ecparam -genkey -name prime256v1 -noout -out ec_private.pem
openssl ec -in ec_private.pem -pubout -out ec_public.pem
gcloud init
gcloud iot devices create test1 --region=us-central1  --registry=itsasecret  --public-key path=ec_public.pem,type=es256
```

the last line adding the device to the registry gives me this error
`ERROR: (gcloud.iot.devices.create) INVALID_ARGUMENT: The device credential in position 0 must be an X.509 certificate.`

Hopefully I'm just missing something simple

datascraps, 2020-08-16:
Update got it working, but I had to piece a bunch of things together that were not laid out anywhere.

Maybe the docs need to be updated?

This is a script I made that works
```
# create registry keys
openssl genpkey -algorithm RSA -out ca_private_registry.pem -pkeyopt rsa_keygen_bits:2048
sudo openssl req -x509 -new -nodes -key ca_private_registry.pem -sha256 -out ca_cert_registry.pem -subj "/CN=unused"
gcloud iot registries credentials create --path=ca_cert_registry.pem  --project=secret  --registry=secret --region=us-central1

# create elipitic device keys
openssl ecparam -genkey -name prime256v1 -noout -out ec_private_device1.pem
sudo openssl req -new -sha256 -key ec_private_device1.pem -out ec_cert_device1.csr -subj "/CN=unused-device"
sudo openssl x509 -req -in ec_cert_device1.csr -CA ca_cert_registry.pem -CAkey ca_private_registry.pem -CAcreateserial -sha256 -out ec_cert_device1.pem
gcloud iot devices create device1 --region=us-central1  --registry=secret  --public-key path=ec_cert_device1.pem,type=es256-x509-pem

# print info to copy to code
openssl ec -in ec_private_device1.pem -noout -text
echo "Copy private part of above to esp8266 code"
```

wasap, 2020-09-27:
thank you. it worked for me too

gguuss, 2020-10-02:
Thank you, I'll update the docs.

closed 2020-10-06 by gguuss

Corne173, 2022-06-25:
I wish someone could explain or refer me to documentation that explains how OpenSSL generates thats 32 byte key derived from the EC private key.  

I'm trying to automate the whole process of adding new devices to the cloud, generating keys for those devices, adding them etc. 
I've used the python `Cryptography` library that's installed with the rest of the google-cloud-api libraries. Documentation found [here](https://cryptography.io/en/latest/hazmat/primitives/key-derivation-functions/).
It generates the private and public EC keys, source documentation found [here](https://cloud.google.com/iot/docs/concepts/device-security?_ga=2.21633339.-1253639980.1654341872&_gac=1.150878404.1654342365.Cj0KCQjwheyUBhD-ARIsAHJNM-NrXnHQX0eOcNSi0gP1KxLCF5VqbIFQFDcAxlKY6rLY8Ue6pzTudXgaAuCcEALw_wcB)
I haven't been able to find much information on how that last part works, that 
`openssl ec -in ec_private_device1.pem -noout -text` part. 

I've used a test platform, found [here](https://github.com/Corne173/IoT_Energy_Logger/tree/master/Tests/ESP%20Connection%20test/Esp8266-lwmqtt).
I am able to connect and publish when I use the key generated using openssl, but not using my implementation.  

    from cryptography.hazmat.primitives.asymmetric import ec
    from cryptography.hazmat.primitives import serialization
    from cryptography.hazmat.primitives import hashes
    from cryptography.hazmat.primitives.kdf.hkdf import HKDF

    private_key = ec.generate_private_key(ec.SECP256R1()) #P-256
    private_pem = private_key.private_bytes(
        encoding=serialization.Encoding.PEM,
        format=serialization.PrivateFormat.PKCS8,
        encryption_algorithm=serialization.NoEncryption()
    )
    with open('../ec_private.pem', 'wb') as filekey:
        filekey.write(private_pem)
    print(private_pem.decode("utf-8"))
    
    public_key = private_key.public_key()
    public_pem = public_key.public_bytes(
       encoding=serialization.Encoding.PEM,
       format=serialization.PublicFormat.SubjectPublicKeyInfo)
    
    with open('../ec_public.pem', 'wb') as filekey:
        filekey.write(public_pem)
    print(public_pem.decode("utf-8"))
    
    
    shared_key = private_key.exchange(
        ec.ECDH(), private_key.public_key())
    # Perform key derivation.
    derived_key = HKDF(
        algorithm=hashes.SHA256(),
        length=32,
        salt=None,
        info=None,
    ).derive(shared_key)
    
    shared_hex_keys = list(map(hex,list(derived_key)))
    with open("../shared_hex_keys.txt", "w") as file:
        data = ",".join(shared_hex_keys)
        file.write(data)
    print(f"Device key:\n {data}")