GoogleCloudPlatform/google-cloud-iot-arduino #189: Gateway is getiing connected again and again ... exactly after 10 seconds

robocircuits, 2020-09-06:
In the loop function
`// This is where the problem is when mqtt calls mqttConnect
    if (!mqttClient->connected()) {
      connect();
    }`
works only when the timeout happens for the MQTTclient.
`mqttClient->setOptions(180, true, 10000); // keepAlive, cleanSession, timeout`

If the timeout is increased resetting of the connection increases.
The situation is getting confusing the more I try to understand it.

gguuss, 2020-10-15:
Please tell us a little more about what's going on: what board are you using, what versions of SSL are installed, and so forth.

robocircuits, 2020-10-16:
Hi
The problem is solved... Its due to timeout...
I was using 12 deligate devices on one gateway. The timeout was 10 sec...
So each delegate device uses 500 milli sec to subscribe so .... It is
talking 12 seconds on the first connection. Hence without subscribing all
it gets disconnected and starts reconnecting.
I increased timeout and decreased delay to 450milli sec

gguuss, 2020-10-16:
I added a note to the readme but it could make sense to add this change to the example.