GoogleCloudPlatform/google-cloud-iot-arduino #191: CO2 Sensor Values Change when I include Esp32 Google Cloud IoT library functions

davidskydvd, 2020-09-16:
I recently made a code that outputs your temp & hum (both through I2C interface) and CO2 (Analog interface) sensors. I'm also using an ESP32 Feather Microcontroller and a ESP32 Featherwing Display (using ILI9341 library) and I am using the Arduino IDE to upload my code to the microncontroller. With the temperature and humidity values already working, the CO2 values is my only concern. Whenever I use the code without the library, the CO2 sensor values are correct, but when I add the library and the GCP Upload function in setup (because I want to upload to GCP), the temp and humidity are fine, but CO2 reads raw values that are too high and as a result, the ppm values are too high.

I have tried a few things and here are my observations:
- Commenting out the setupIoTCloud() function results in giving correct readings for all three sensors (Temp, Hum, CO2)
- Whenever I add the setupIoTCloud() function, it messes up my CO2 ppm readings taken from analog readings. However, Temperature and humidity readings that use up the ESP32's I2C interface are outputting correct values.

I wish to know what's going on in the library that's causing the CO2 values that are reading from the Analog interface to maximize.

Below is my code with the library and its functions included.

```

//CO2 TEMP HUM GCP TEST 

#include "esp32-mqtt.h"
#include <Wire.h>
#include "Adafruit_SHT31.h"
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>

#ifdef ESP32
  #define STMPE_CS  32
  #define TFT_CS    15
  #define TFT_DC    33
  #define SD_CS     14
#endif

#define window_size 50

//AQMS VALUES:
// temp & hum:
float t = 0;
float h = 0; 

float c = 0; // CO2
//DATA SMOOTHING VARIABLES:
int indices = 0;
int sensorValue = 0;
int sum = 0;
int readings[window_size];
int averaged = 0;
byte sensorIn = A0;

//WCS VALUES: (no readings)
float p = 0; //ph
float e = 0; //ec
float w = 0; //watertemp

String payload;

String sys1 = "AQMS0111";
String devID1 = "espDDSTM";
int b = 7;
String tant1 = "IBC";
String sit1 = "Mabolo";

int voltage_diference;
float voltage;

//Intervals:

//CO2 SENSOR:
unsigned long co2MillisSet = 0;
unsigned long co2MillisInterval = 1000; //1s

//DISPLAY:
unsigned long printMillisSet = 0;
unsigned long printMillisInterval = 1000; //1s

unsigned long temphumMillisSet = 0;
unsigned long temphumMillisInterval = 1000;

//GCP:
unsigned long uploadMillis = 0;
unsigned long uploadMillisInterval = 150000; //2.5mins
//INTERNET RECONNECTION:
unsigned long check_wifi = 30000; // 30s

uint8_t rotation = 1;

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
Adafruit_SHT31 sht31 = Adafruit_SHT31();

void setup(){
  Serial.begin(115200);
  analogReadResolution(12);
  
  tft.begin();
  tft.setRotation(rotation);
  
  if (!sht31.begin(0x44)) {
    Serial.println("Error reading SHT31");
  }
 
  tft.fillScreen(ILI9341_GREEN);
  tft.setCursor(0,50);
  tft.setTextColor(ILI9341_WHITE);
  tft.setTextSize(3);
  tft.print("NXTLVL AQMS0111");
  delay(3000);

  // Read diagnostics (optional but can help debug problems)
  uint8_t x = tft.readcommand8(ILI9341_RDMODE);
  Serial.print("Display Power Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDMADCTL);
  Serial.print("MADCTL Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDPIXFMT);
  Serial.print("Pixel Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDIMGFMT);
  Serial.print("Image Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDSELFDIAG);
  Serial.print("Self Diagnostic: 0x"); Serial.println(x, HEX);

  setupCloudIoT(); // <-------- what messes up the CO2 analog readings, uncomment it and everything works, except no upload 
                  //            to GCP.
  
}

void loop(){
  unsigned long universalMillis = millis();
  
  mqtt->loop();
  delay(10);

  if(!mqttClient->connected()){
  connect();  
  }

  if(universalMillis - temphumMillisSet > temphumMillisInterval){
  t = sht31.readTemperature();
  h = sht31.readHumidity();
  }
  readCO2(); // Returns c (carbon dioxide)

  printValues(t, h, c); 
  
 if(millis() - uploadMillis > uploadMillisInterval){
 payload   =  String("{\"System\":") + ("\"") + sys1  + ("\"") +   //string
                     String(",\"deviceID\":") + ("\"") + devID1 + ("\"") + //string
                     String(",\"Box\":") + ("\"") + b + ("\"") + //integer
                     String(",\"tank\":") + ("\"") + tant1 + ("\"") + //string
                     String(",\"site\":") + ("\"") + sit1 + ("\"") + //string
                     String(",\"timecollected\":") + ("\"") + time(nullptr) + ("\"") + //timestamp
                     String(",\"pH\":") + ("\"") + p + ("\"") + //float
                     String(",\"EC\":") + ("\"") + e + ("\"") + //float
                     String(",\"waterTemp\":") + ("\"") + w + ("\"") + //float
                     String(",\"airTemp\":") + ("\"") + t + ("\"") + //float
                     String(",\"airHum\":") + ("\"") + h + ("\"") + //float
                     String(",\"CO2\":") + ("\"") + c + ("\"") + //float
                     String("}"); 
 
 Serial.print(payload); 
 publishTelemetry(payload); 

 uploadMillis = millis();
 }
 
}

void readCO2(){
//Read voltage
  if(millis() - co2MillisSet > co2MillisInterval){
  co2MillisSet = millis();
  sensorValue = ReadVoltage(sensorIn) * 1000;
  sum = sum - readings[indices];
  readings[indices] = sensorValue;
  sum = sum + sensorValue;
  indices = (indices + 1) % window_size;

  averaged = sum / window_size;
  
  // The analog signal is converted to a voltage
  voltage = averaged*(3300/4096.0);
  if(voltage == 0)
  {
    Serial.println("Fault");
  }
  else if(voltage < 400)
  {
    Serial.println("preheating");
  }
  else
  {
    voltage_diference=voltage-300;
    c = voltage_diference*50.0/16.0;
    // Print Voltage
    Serial.print("Raw Analog Values: ");
    Serial.println(sensorValue);
    Serial.print("Average: ");
    Serial.println(averaged);
    Serial.print("voltage: ");
    Serial.print(voltage);
    Serial.println("mv");
    //Print CO2 concentration
    Serial.print(c);
    Serial.println("ppm");

  }
 }
}

void printValues(float temp, float hum,float carbondioxide){
if(millis() - printMillisSet > printMillisInterval){
 tft.fillScreen(ILI9341_BLACK);
 tft.setCursor(0,50);
 tft.setTextColor(ILI9341_WHITE);
 tft.setTextSize(3);

 if (!isnan(temp)){
    tft.print("Temp : "); tft.println(temp);
    Serial.print("airTemp : "); Serial.println(temp);
  }
  else {
    tft.print("Temp : "); tft.println("ERROR");
    Serial.println("Error reading SHT31-temperature");
  }
  if (!isnan(hum)) {
    tft.print("Hum : "); tft.println(hum);
    Serial.print("airHum : "); Serial.println(hum);
  }
  else {
    tft.print("Hum : "); tft.println("ERROR");
    Serial.print("Error reading SHT31-humidity");
  }
  tft.print("CO2 : "); tft.println(carbondioxide);

 printMillisSet = millis();
 }
}

double ReadVoltage(byte pin){
  double reading = analogRead(pin); // Reference voltage is 3v3 so maximum reading is 3v3 = 4095 in range 0 to 4095
  if(reading < 1 || reading > 4095) return 0;
  //return -0.000000000009824 * pow(reading,3) + 0.000000016557283 * pow(reading,2) + 0.000854596860691 * reading + 0.065440348345433;
   return -0.000000000000016 * pow(reading,4) + 0.000000000118171 * pow(reading,3)- 0.000000301211691 * pow(reading,2)+ 0.001109019271794 * reading + 0.034143524634089;
} // Added an improved polynomial, use either, comment out as required 

```

Here are results of my code with and without the Google IoT Core library:

> First Code: (No Google CLoud  IoT core library)
- 23:49:23.695 -> Raw Analog Values: 510
- 23:49:23.695 -> Average: 525
- 23:49:23.695 -> voltage: 456.72mv
- 23:49:23.695 -> 417.26ppm

> Second Code: (w/ Google Cloud IoT Core library)
- 23:49:23.695 -> Raw Analog Values: 3140
- 23:49:23.695 -> Average: 3140
- 23:49:23.695 -> voltage: 2529.79mv
- 23:49:23.695 -> 6965.63ppm

> Note: CO2 values change accordingly with the first code, but second code always maximizes to 6965.63ppm without regard to the environment.

Thank you and stay safe,
David

gguuss, 2020-10-15:
My best guess is that this is a threading issue on the ESP32 and this library is not the best place to support this. Because you're using an ESP32, you should be pinning a task for updating the sensor readings. 

Example code for creating a ESP32 task:

```cpp
  // Start task to get temperature
  xTaskCreatePinnedToCore(
      tempTask,                       /* Function to implement the task */
      "tempTask ",                    /* Name of the task */
      4000,                           /* Stack size in words */
      NULL,                           /* Task input parameter */
      5,                              /* Priority of the task */
      &tempTaskHandle,                /* Task handle. */
      1);                             /* Core where the task should run */
```

You can find a [full example of this pattern in this GH](https://github.com/beegee-tokyo/DHTesp) but I'm pretty sure your issue is something outside of this library.

closed 2020-10-15 by gguuss

davidskydvd, 2020-10-16:
Thank you, will look into this!

David Sta. Maria
NXTLVL Farms Inc.