GoogleCloudPlatform/google-cloud-iot-arduino #204: Stability Connection Problem

jbarriosmuriel, 2020-10-21:
Hi! Congratulation for the work.

I'm using this library to connect a ESP32 module. I send 100 data per seconds and and sometimes I detect stability problems.
There is pause in the connection randomly interrupting the data flow and communication.
On the other hand, I have a client conect to de topic to read the data (it appears almost in real time). I would like to know what are the possible causes of the connection pauses / resumes problem.

Thanks very much for your responses

gguuss, 2020-11-11:
100 Updates per second seems pretty high frequency. Note that the [limit is 100 messages / second](https://cloud.google.com/iot/quotas) so you are right at the maximum, you may want to coalesce your payloads.

My guess on the pausing that you are experiencing is happening at a low level on the FreeRTOS level. You may want to try [dedicating the processing in a specific core using xTaskCreatePinnedToCore](https://techtutorialsx.com/2017/05/09/esp32-running-code-on-a-specific-core/) but this is not a supported scenario for this library.

jbarriosmuriel, 2020-11-12:
Thanks for the reply. Sorry I explained wrong, it is 100 data / seconds but they are sent in packets, a total of 10 per second.
Still the stability issues are random. I am not able to fix them