GoogleCloudPlatform/google-cloud-iot-arduino #208: Connection not stable.

josemtrg, 2020-11-09:
Im trying to implement the given example for a NodeMCU using this example:

https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Esp8266-lwmqtt

Everything seems to work but the problem is that the connection is not stable. It connect and disconnects right after a few seconds. This is the logs i get:

///////////////////////////////
Connecting to WiFi
Waiting on time sync...
Connecting...
Refreshing JWT
connected

Library connected!
0
 Connecting...
Refreshing JWT
connected

Library connected!
0
 Connecting...
Refreshing JWT
connected
////////////////////////////////

It seems that the problem is on the mqtt loop function

gguuss, 2020-11-09:
Did you make any changes to the example? If you use the MQTT protocol in a way the bridge does not expect, ex. publish to the wrong topic, the protocol bridge will disconnect you.

josemtrg, 2020-11-10:
@gguuss Thanks. That was it. I was missing the "events" topic.

closed 2020-11-10 by josemtrg