GoogleCloudPlatform/google-cloud-iot-arduino #210: How to have ESP8266 subscribe to PubSub Topic? [Easy win]

Steven-N-Hart, 2020-11-19:
Awesome work everyone. I do have a naive question that would be useful to cover in the examples though. All the examples show how to publish data to a topic, but none show how to have the device receive feedback. For example, say I have a function that turns on and off an LED every minute. I'd like to send a message to the device to stop blinking altogether. Reading the documentation [here](https://cloud.google.com/iot/docs/how-tos/commands#iot-core-send-command-python), it seems as though a command needs to be sent (or a configuration). Either way, one could parse out the message to turn off the light. However, the documentation on cloud.google.com does not include the Arduino code that would need to be embedded into the device so it could subscribe/read/parse appropriately.

**Could someone post here or submit a PR that shows the recommended way for setting up this type of functionality? I believe it would be very useful to Embedded C noobs like myself.**

Many thanks if you are available to help. I know we all have a lot on our plates, but I think this would be incredibly useful and would require minimal effort from one of the experts.

gguuss, 2020-11-19:
In all of the examples, t[he callback function](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp8266-lwmqtt/esp8266_mqtt.h#L36-L44) is triggered on any message (command/configuration) sent to a device. Any functionality you want to add can be added there, for example, in the camera example I think the command topic is used to trigger taking a photo.

Steven-N-Hart, 2020-11-19:
Thanks @gguuss, but I guess what I'm getting at is _how_ to send a message to the device. Can it subscribe to a topic, or does a function need to exist entirely outside of the device.  I'm not quite clear on how to use the callback completely.

Steven-N-Hart, 2020-11-19:
Update: If you use the example provided, it's already hooked up to receive the message by default.  There isn't anything more on the device that needs to be done to receive commands. To actually send message, one needs to send it from an external source (like the [console](https://cloud.google.com/iot/docs/how-tos/commands#iot-core-send-command-python). Hopefully this helps others!

closed 2020-11-19 by Steven-N-Hart