GoogleCloudPlatform/google-cloud-iot-arduino #215: Narrowing conversion of '2024104815l' from 'long int' to 'WORD

JuniorEC, 2021-01-15:
I'm trying to use the library with TinyGsm and I'm taking this strange error when I compile

Board: Arduino Mega 2560

Code:
```
#include <Wire.h>

#include <time.h>
#include <Arduino.h>

#include <MQTT.h>

#include <CloudIoTCore.h>
#include <CloudIoTCoreMqtt.h>
#include "ciotc_config.h" // Wifi configuration here

#define TINY_GSM_MODEM_SIM800
#include <TinyGsmClient.h>
#include <SoftwareSerial.h>

#define SerialMon Serial
#define GSM_AUTOBAUD_MIN 9600
#define GSM_AUTOBAUD_MAX 9600
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
///////////////////////////////
const char apn[] = "internet";
const char gprsUser[] = "";
const char gprsPass[] = "";
SoftwareSerial SerialAT(10, 11); // RX, TX
TinyGsm modem(SerialAT);
TinyGsmClientSecure *gsmclient;

// Initialize WiFi and MQTT for this board
MQTTClient *mqttClient;
// BearSSL::WiFiClientSecure *netClient;

CloudIoTCoreDevice *device;
CloudIoTCoreMqtt *mqtt;
unsigned long iat = 0;
String jwt;

String getJwt()
{
  // Disable software watchdog as these operations can take a while.
  // ESP.wdtDisable();
  iat = time(nullptr);
  Serial.println("Refreshing JWT");
  jwt = device->createJWT(iat, jwt_exp_secs);
  // ESP.wdtEnable(0);
  return jwt;
}

void connect()
{
  mqtt->mqttConnect();
}

// TODO: fix globals
void setupCloudIoT()
{
  // Create the device
  device = new CloudIoTCoreDevice(
      project_id, location, registry_id, device_id,
      private_key_str);

  // ESP8266 WiFi setup
  //gsmclient = gsmclient(modem);
  gsmclient = new TinyGsmClientSecure(modem);
[... 211 lines cut ...]
 #define EP0(x) (ROTRIGHT(x,2) ^ ROTRIGHT(x,13) ^ ROTRIGHT(x,22))
                 ^
.pio\libdeps\megaatmega2560\Google Cloud IoT Core JWT\src\crypto\sha256.cpp:138:7: note: in expansion of macro 'EP0'
  t2 = EP0(a) + MAJ(a,b,c);
       ^
.pio\libdeps\megaatmega2560\Google Cloud IoT Core JWT\src\crypto\sha256.cpp:26:54: warning: left shift count >= width of type [-Wshift-count-overflow]
 #define ROTRIGHT(a,b) (((a) >> (b)) | ((a) << (32-(b))))
                                                      ^
.pio\libdeps\megaatmega2560\Google Cloud IoT Core JWT\src\crypto\sha256.cpp:30:33: note: in expansion of macro 'ROTRIGHT'
 #define EP0(x) (ROTRIGHT(x,2) ^ ROTRIGHT(x,13) ^ ROTRIGHT(x,22))
                                 ^
.pio\libdeps\megaatmega2560\Google Cloud IoT Core JWT\src\crypto\sha256.cpp:138:7: note: in expansion of macro 'EP0'
  t2 = EP0(a) + MAJ(a,b,c);
       ^
.pio\libdeps\megaatmega2560\Google Cloud IoT Core JWT\src\crypto\sha256.cpp:26:34: warning: right shift count >= width of type [-Wshift-count-overflow]
 #define ROTRIGHT(a,b) (((a) >> (b)) | ((a) << (32-(b))))
                                  ^
.pio\libdeps\megaatmega2560\Google Cloud IoT Core JWT\src\crypto\sha256.cpp:30:50: note: in expansion of macro 'ROTRIGHT'
 #define EP0(x) (ROTRIGHT(x,2) ^ ROTRIGHT(x,13) ^ ROTRIGHT(x,22))
                                                  ^
.pio\libdeps\megaatmega2560\Google Cloud IoT Core JWT\src\crypto\sha256.cpp:138:7: note: in expansion of macro 'EP0'
  t2 = EP0(a) + MAJ(a,b,c);
       ^
*** [.pio\build\megaatmega2560\lib5ae\Google Cloud IoT Core JWT\crypto\sha256.cpp.o] Error 1
```

SamGeens, 2021-02-19:
Hi

Have you found the solution for your problem yet? 

I'm trying to use this library just like you on an Arduino mega 2560 but I'm using WifiESP with a ESP8266 shield. I'm able to make a wifi connection but I'm having problems with generating the JWT token. I've been debugging for a while now and it seems that the JWT generation isn't working because of the public key is loaded in `static BearSSL::X509List certList;` which you removed aswell from the code. This might be the (one of the) explaination(s) for your error. 

It might be that this library would be able to generate ES256 jwt tokens on AVR architecture with some minor changes. Anyone who can help with this?

gguuss, 2021-02-19:
> It might be that this library would be able to generate ES256 jwt tokens on AVR architecture with some minor changes. Anyone who can help with this?

I'm able to give some limited support but am not actively developing the library. If you need code review and change verification, happy to review and test PRs.

SamGeens, 2021-02-20:
I've made some small changes to make the library run on arduino uno and mega. But when generating the JWT token it gets stuck in an endless loop on AVR architecture. The problem is in `ecc_win_mul` in ecc.cpp and is probably because the public key is not loaded because of the `certList.append(primary_ca)` line is removed. This had to be removed because BearSSL does not support AVR architecture. 

Do you perhaps have any idea on how to load the key correctly without BearSSL? Can it be possible as string parameter just like the private key (`device.setPrivateKey(private_key);`)?  

I know it is bad habbit but an other solution (at least for testing) could be to skip the verify part because that the only use for the public key?