GoogleCloudPlatform/google-cloud-iot-arduino #220: weird disconnecting behaviour !

warshaa, 2021-03-19:
Hi

I've used the Library few months back on my esp32 and it worked perfectly without issues.

however now I'm facing some connections problem : 

usually it will not connect from the first time and I keep getting the message : 

```
Refreshing JWT
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE2MTYxNTY3MTUsImV4cCI6MTYxNjE1NzkxNSwiYXVkIjoiZXNwODI2Ni0yOTg4MTYifQ.kTr058Mms_PklVOsXT3fT4jTUeiccQcvUEmnm3PAxeJS_cG9JNz98vS4IXerQSbH0mkAsJU-8jEhpm6AUO80JQ
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/esp8266-298816/locations/europe-west1/registries/xxx/devices/xxx1
Waiting 60 seconds, retry will likely fail
```

then it successfully connects after few attempts and works fine and again it will disconnect after some time and then connects after many attempts ... and so on

it is completely random so there is no pattern and I'm trying just to send a const string

Goolge logs errors :

[9] mqtt: The connection broke or was closed by the client.
or  just [9]

the same code was working fine few months back the only things changed are my location and the esp32 board 

```
`[env:esp32dev]`
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = 
	googlecloudplatform/Google Cloud IoT Core JWT@^1.1.11
	256dpi/MQTT@^2.5.0
	circuitsetup/ATM90E32@^1.1.0
	bblanchon/ArduinoJson@^6.17.2
```

abliveira, 2021-04-05:
Same issue here. I've tried with new and old projects (which worked before) and could not get it working. Is this system now broken?

raphael-bmec-co, 2021-04-05:
Have a look at #144. May well be related to the issues mentioned there arising from the token being stored as a string which may contain the end of string character.

warshaa, 2021-04-20:
I haven't tried for a while , but something I found out it might be esp32 power issue.
my laptop usb current vary depends on the battery level charge , so when the battery level is low the current is low , the esp32 will still work with no brownout or restart however the wifi connection will drop which causes the disconnecting problem
if someone is facing similar problem make sure that ur usb sources efficient current or/and adding a decoupling capacitors might help too

raphael-bmec-co, 2021-04-21:
@warshaa we have also observed this behaviour with respect to the Wi-Fi going down and staying down.

aaryaapg, 2021-06-04:
I'm using ESP32, and I'm facing the exact same issue.  
```
Settings incorrect or missing a cyper for SSL 
Connect with mqtt.2030.ltsapis.goog:8883
```
I'm certain that the device ID and every other detail is correct. Were you able to fix it ?