GoogleCloudPlatform/google-cloud-iot-arduino #222: Cannot generate JWT on ESP8266

lptr, 2021-04-13:
My code works well on a LilyGo ESP32 board, but when I try to run it using an ESP8266 board, I always get a crash like this:

```text
01:01:48.224 > Initializing MQTT connector...
01:02:18.219 > Current time is 1618354938
01:02:18.242 > Connecting...
01:02:18.244 > Refreshing JWT
01:02:19.941 > 
01:02:19.941 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
01:02:19.947 > 
01:02:19.947 > Soft WDT reset
01:02:19.948 > 
01:02:19.949 > >>>stack>>>
...
01:02:20.470 > <<<stack<<<
01:02:20.470 > 
01:02:20.470 > 0x4020e0da in NN_AddDigitMult(unsigned int*, unsigned int*, unsigned int, unsigned int*, unsigned char) at ??:?
01:02:20.470 > 0x4020e184 in NN_Mult(unsigned int*, unsigned int*, unsigned int*, unsigned char) at ??:?
01:02:20.470 > 0x4020e0da in NN_AddDigitMult(unsigned int*, unsigned int*, unsigned int, unsigned int*, unsigned char) at ??:?
01:02:20.470 > 0x4020e725 in NN_ModInv(unsigned int*, unsigned int*, unsigned int*, unsigned char) at ??:?
01:02:20.470 > 0x4020d8a5 in ecc_add(point*, point*, point*) at ??:?
01:02:20.470 > 0x4020d954 in ecc_win_precompute(point*, point*) at ??:?
01:02:20.470 > 0x4020de1c in ecdsa_init(point*) at ??:?
01:02:20.470 > 0x4020d2b8 in CreateJwt(char const*, long long, unsigned int*, int) at ??:?
01:02:20.470 > 0x40101044 in realloc at ??:?
01:02:20.470 > 0x40100e71 in realloc at ??:?
01:02:20.470 > 0x40218db4 in uart_do_write_char at uart.cpp:?
01:02:20.470 > 0x40215718 in HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
01:02:20.470 > 0x40100b6b in umm_free_core at umm_malloc.cpp:?
01:02:20.470 > 0x40100622 in millis at ??:?
01:02:20.470 > 0x4020c69e in CloudIoTCoreDevice::createJWT(long long, int) at ??:?
01:02:20.470 > 0x40215718 in HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
01:02:20.470 > 0x40215aa5 in Print::write(char const*) at ??:?
01:02:20.470 > 0x40215d24 in Print::println() at ??:?
01:02:20.470 > 0x40208b7b in MqttHandler::getJwt() at ??:?
01:02:20.470 > 0x40208ba8 in getJwt() at ??:?
01:02:20.470 > 0x4020ceec in CloudIoTCoreMqtt::mqttConnect(bool) at ??:?
01:02:20.470 > 0x40223082 in strdup at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/string/strdup.c:11
01:02:20.470 > 0x4020c385 in MQTTClient::onMessage(void (*)(String&, String&)) at ??:?
01:02:20.470 > 0x40209b39 in MqttHandler::begin(Client&, ArduinoJson6173_91::JsonDocument const&, std::function<void (ArduinoJson6173_91::JsonDocument&)>, std::function<void (ArduinoJson6173_91::JsonDocument&)>) at ??:?
01:02:20.470 > 0x4020ae22 in setup at ??:?
01:02:20.470 > 0x4020a1a4 in std::_Function_base::_Base_manager<setup::{lambda(ArduinoJson6173_91::JsonDocument const&)#2}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda(ArduinoJson6173_91::JsonDocument const&)#2}> const&, std::_Manager_operation) at main.cpp:?
01:02:20.470 > 0x4020a74c in std::_Function_handler<void (ArduinoJson6173_91::JsonDocument&), setup::{lambda(ArduinoJson6173_91::JsonDocument const&)#2}>::_M_invoke(std::_Any_data const&, ArduinoJson6173_91::JsonDocument&) at main.cpp:?
01:02:20.470 > 0x40220268 in esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::findCachedDomainOffset(void const*, bool) const at ??:?
01:02:20.470 > 0x40209e02 in std::_Function_base::_Base_manager<OtaHandler::begin(char const*)::{lambda(ota_error_t)#4}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<OtaHandler::begin(char const*)::{lambda(ota_error_t)#4}> const&, std::_Manager_operation) at OtaHandler.cpp:?
01:02:20.470 > 0x4020a170 in std::_Function_base::_Base_manager<setup::{lambda(ArduinoJson6173_91::JsonDocument const&)#1}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda(ArduinoJson6173_91::JsonDocument const&)#1}> const&, std::_Manager_operation) at main.cpp:?
01:02:20.470 > 0x4020a148 in std::_Function_handler<void (ArduinoJson6173_91::JsonDocument&), setup::{lambda(ArduinoJson6173_91::JsonDocument const&)#1}>::_M_invoke(std::_Any_data const&, ArduinoJson6173_91::JsonDocument&) at main.cpp:?
01:02:20.470 > 0x402175c8 in loop_wrapper() at core_esp8266_main.cpp:?
01:02:20.470 > 0x401011d5 in cont_wrapper at ??:?
01:02:20.470 > 
01:02:20.470 > 
01:02:20.470 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

The actual stack under `CreateJwt()` is always a bit different, this is another variation:

```text
01:03:29.219 > 0x4020ea10 in omega_mul(unsigned int*, unsigned int*, unsigned int*, unsigned char) at ??:?
01:03:29.219 > 0x4020e3a8 in NN_ModSqrOpt(unsigned int*, unsigned int*, unsigned int*, unsigned int*, unsigned char) at ??:?
01:03:29.219 > 0x4020d78d in c_add_mix(point*, unsigned int*, point*, unsigned int*, point*) at ecc.cpp:?
01:03:29.219 > 0x4020dc9c in ecc_win_mul(point*, unsigned int*, point*) at ??:?
01:03:29.219 > 0x4020dd60 in ecc_gen_pub_key(unsigned int*, point*) at ??:?
01:03:29.219 > 0x4020d2b2 in CreateJwt(char const*, long long, unsigned int*, int) at ??:?
01:03:29.219 > 0x40101044 in realloc at ??:?
01:03:29.219 > 0x40100e71 in realloc at ??:?
01:03:29.219 > 0x40218db4 in uart_do_write_char at uart.cpp:?
01:03:29.219 > 0x40215718 in HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
01:03:29.219 > 0x40100b6b in umm_free_core at umm_malloc.cpp:?
01:03:29.219 > 0x40100622 in millis at ??:?
01:03:29.219 > 0x4020c69e in CloudIoTCoreDevice::createJWT(long long, int) at ??:?
```

raphael-bmec-co, 2021-04-14:
see #144

lptr, 2021-04-14:
> see #144

@raphael-bmec-co what part of that issue do you refer to? I don't see any report about a crash there. What am I missing?

raphael-bmec-co, 2021-04-15:
@lptr apologies, I misread your issue. I did not see that you had it working on another micro.