GoogleCloudPlatform/google-cloud-iot-arduino #224: publishTelemetry doesn't send data when separate status topic enabled in registry

sacculus, 2021-06-03:
Hi
I've got strange behavior - when added separate state topic to registry, my device stopped publishing telemetry.
I check return code and it show that publishing is OK, but haven't see any message in device-events (default telemetry) topic.
Also in according to serial log of device it sends telemetry but I've not see any records about it in log (debug level enabled) at gcloud console...
I tried to swap order of state and telemetry publishing in code but no success.
I'll try to remove state topick and check sending telemetry again. But I think that looks like a bug in case if I didn't something wrong in my code.

Any help, please!

This is code of publishing:

  /*
   * Send device state
   */
  Serial.print("State: ");
  String state = getDeviceState();
  Serial.println(state);
  Serial.print("Publishing state... ");
  if (publishState(state)) {
    Serial.println("done");
  } else {
    Serial.print("error: ");
    mqtt->logError();
  }
  /*
   * Send device telemetry
   */
  Serial.print("Telemetry: ");
  String report = getDefaultReport();
  Serial.println(report);
  Serial.print("Publishing telemetry... ");
  if (publishTelemetry(report)) {
    Serial.println("done");
  } else {
    Serial.print("error: ");
    mqtt->logError();
  }

and this is log from serial console:

09:54:39.343 -> State: {"device_id":"shit_meter-898da8","chip_model":"ESP32-D0WDQ6","chip_revision":1,"chip_cores":2,"firmware_version":"0.0.1","ssid":"wl-500gpv2","bssid":"14:CC:20:62:C9:8F"}
09:54:39.343 -> Publishing state... done
09:54:39.343 -> Telemetry: {"battery":3294,"temperature":2241,"distance":221}
09:54:39.641 -> Publishing telemetry... done

sacculus, 2021-06-03:
Update
Small mistake in previous post. When I changed publishing order when telemetry messages were sent first it were delivered but there was some lag in cloud console to show them.

Also I did small set of tests.
After remove separate state topic from registry telemetry messages began to be delivered.
Then I reverted back state topic into registry and also swaped publishing order to send state first and telemetry second - and telemetry wasn't delivered again.
At end I swapped order of publishing telemetry end state messages back - telemetry messages began to delivered.

So, in case when defined two topics (events and states) in registry it works only when publishing of telemetry goes first.
It is very strange for me...

gguuss, 2021-06-03:
The topic for a your device must be the device-specific topic `projects/PROJECT_ID/locations/REGION/registries/REGISTRY_ID/devices/DEVICE_ID` but I'm guessing you're doing this correctly if you are using the Arduino library and are using the global functions as I'm seeing in your example.  That said, if the topic is incorrect, the MQTT device bridge will disconnect the client which creates inconsistent behavior in the client. If you have enabled Cloud Logging for your device registry, additional information may be available in the console.

Messages published to the events telemetry topic can have some latency before being available to the PubSub queue, particularly when the first messages get published (I'm not sure I've seen more than 60 seconds in practice, which is rare, and I've used the MQTT bridge quite a bit). The QoS can affect the latency so it's worth trying both 0 and 1 and seeing which works better for your application. If you

State messages persist (you can read the last 10 states for a device) and are easier for testing because you can see them on the [Google Cloud console](https://console.cloud.google.com/iot). 

[Full details on the MQTT bridge are documented here](https://cloud.google.com/iot/docs/how-tos/mqtt-bridge) and may be helpful for debugging.

sacculus, 2021-06-08:
Hi
I suspect that it is not related to delay in cloud.
My opinion is that this behaviour is more related to some asyncronous execution in mqtt (possible) library or somesing wrong with processing of state of sending data.
For example I'll show next case
(Small remark: sometime, very rare I've got case when both messages were not delivered)

Log on device shows that both publications - events&state, where done with succes. There is it:

15:17:21.422 -> WiFi connected
15:17:21.422 -> Synchronizing time... done
15:17:21.422 -> Initializing Google Cloud IoT...done
15:17:21.422 -> Connecting...
15:17:21.422 -> Refreshing JWT
15:17:24.675 -> connected
15:17:24.675 -> 
15:17:24.675 -> Library connected!
15:17:24.741 -> incoming: /devices/shit_meter-898da8/config:
15:17:24.741 -> {"duty_cycle_interval_sec":1800}
15:17:24.741 -> No more task for GCloud, can go asleep
15:17:24.808 -> Telemetry: {"battery":3282,"temperature":2367,"distance":220}
15:17:25.140 -> Publishing telemetry... done
15:17:25.140 -> State: {"device_id":"shit_meter-898da8","chip_model":"ESP32-D0WDQ6","chip_revision":1,"chip_cores":2,"firmware_version":"0.0.3","ssid":"wl-500gpv2","bssid":"14:CC:20:62:C9:8F"}
15:17:25.140 -> Publishing state... done
15:17:25.140 -> State changed from: 3 to 4
15:17:25.638 -> Close connection to cloud... done
15:17:25.804 -> System going to sleep for 1800 seconds

But log in cloud shows that only one of publications was received - events. And moreover, device was disconnected because timeout error after more than 1 minute of waiting, so it was enough time to delivery second message.
Cloud log at attached screen picture.
[image]

In my code I control return code of both publishing functions. And write error message to serial console if something goes wrong.
For reliable finish of all tasks when all are done I also wait half of second with nonblocking counter at loop().
There is code:

bool publishTelemetry(String data){
  uint8_t retry = 0;
  bool result = false;

  while (retry++ < p_gcloud_publishing_retries.value()) {
    if ((result = mqtt->publishTelemetry(data))) {
      break;
    }
    Serial.print("error: ");
    mqtt->logError();
    if (retry < p_gcloud_publishing_retries.value()) {
      delay(GCLOUD_PUBLISHING_RETRY_DELAY_MSEC);
      Serial.print("Retrying... ");
    }
  }
  return result;
}

bool publishState(String data) {
  uint8_t retry = 0;
  bool result = false;

  while (retry++ < p_gcloud_publishing_retries.value()) {
    if ((result = mqtt->publishState(data))) {
      break;
    }
    Serial.print("error: ");
    mqtt->logError();
    if (retry < p_gcloud_publishing_retries.value()) {
      delay(GCLOUD_PUBLISHING_RETRY_DELAY_MSEC);
      Serial.print("Retrying... ");
    }
  }
  return result;
}

// Cut begin: publishing messages
  Serial.print("Telemetry: ");
  String report = getDefaultReport();
  Serial.println(report);
  Serial.print("Publishing telemetry... ");
  if (publishTelemetry(report)) {
    Serial.println("done");
  } else {
    Serial.println("Fatal error when publishing, system will restarted...");
    Serial.print("Close connection to cloud... ");
    if (mqttClient->connected()) {
      mqttClient->disconnect();
      delay(100);
    }
    Serial.println("done");
    Serial.println("Rebooting");
    ESP.restart();
  }

  Serial.print("State: ");
  String state = getDeviceState();
  Serial.println(state);
  Serial.print("Publishing state... ");
  if (publishState(state)) {
    Serial.println("done");
  } else {
    Serial.println("Fatal error when publishing, system will restarted...");
    Serial.print("Close connection to cloud... ");
    if (mqttClient->connected()) {
      mqttClient->disconnect();
      delay(100);
    }
    Serial.println("done");
    Serial.println("Rebooting");
    ESP.restart();
  }
// Cut end: publishing messages

// Cut begin: finish job
  if (systemCanGoAsleep && gcloudCanGoAsleep && !systemConfigChangesPending) {
    if (sleepDeadline == 0) {
      sleepDeadline = millis() + SLEEP_DELAY_MSEC;
    } else {
      if (sleepDeadline <= millis()) {
        goToSleep(p_duty_cycle_interval_sec.value());
      }
    }
  }
// Cut end: finish job

sacculus, 2021-06-08:
As only I've wrote previous post I cought another issue...

Two times with three retries publishing was ended with error.
But receiving message on configuration topic was successed all times, so there eliminates possible networking issues.

It is noteworthy that device log shows that publication was successful at last iteration, but cloud log shows that messages were not received.

Log of device (slightly truncated):

16:17:10.023 -> ets Jun  8 2016 00:22:57
... skip ...
16:17:12.744 -> Initializing Google Cloud IoT...done
16:17:12.744 -> Connecting...
16:17:12.744 -> Refreshing JWT
16:17:15.997 -> connected
16:17:15.997 -> 
16:17:15.997 -> Library connected!
16:17:16.063 -> incoming: /devices/shit_meter-898da8/config:
16:17:16.063 -> {"duty_cycle_interval_sec":1800}
16:17:16.063 -> No more task for GCloud, can go asleep
16:17:16.395 -> Telemetry: {"battery":3277,"temperature":2455,"distance":220}
16:17:16.694 -> Publishing telemetry... error: -9
16:17:16.694 -> LWMQTT_MISSING_OR_WRONG_PACKET
16:17:17.192 -> Retrying... error: -9
16:17:17.192 -> LWMQTT_MISSING_OR_WRONG_PACKET
16:17:17.723 -> Retrying... error: -9
16:17:17.723 -> LWMQTT_MISSING_OR_WRONG_PACKET
16:17:17.723 -> Fatal error when publishing, system will restarted...
16:17:17.723 -> Close connection to cloud... done
16:17:17.723 -> Rebooting

16:17:17.756 -> ets Jun  8 2016 00:22:57
... skip ...
16:17:20.545 -> Initializing Google Cloud IoT...done
16:17:20.545 -> Connecting...
16:17:20.545 -> Refreshing JWT
16:17:23.764 -> connected
16:17:23.764 -> 
16:17:23.764 -> Library connected!
16:17:23.831 -> incoming: /devices/shit_meter-898da8/config:
16:17:23.831 -> {"duty_cycle_interval_sec":1800}
16:17:23.831 -> No more task for GCloud, can go asleep
16:17:24.196 -> Telemetry: {"battery":3270,"temperature":2453,"distance":220}
16:17:24.495 -> Publishing telemetry... error: -9
16:17:24.495 -> LWMQTT_MISSING_OR_WRONG_PACKET
16:17:24.993 -> Retrying... error: -9
16:17:24.993 -> LWMQTT_MISSING_OR_WRONG_PACKET
16:17:25.491 -> Retrying... error: -9
16:17:25.491 -> LWMQTT_MISSING_OR_WRONG_PACKET
16:17:25.491 -> Fatal error when publishing, system will restarted...
16:17:25.491 -> Close connection to cloud... done
16:17:25.491 -> Rebooting

16:17:25.557 -> ets Jun  8 2016 00:22:57
... skip ...
16:17:28.379 -> Initializing Google Cloud IoT...done
16:17:28.379 -> Connecting...
16:17:28.379 -> Refreshing JWT
16:17:31.599 -> connected
16:17:31.599 -> 
16:17:31.599 -> Library connected!
16:17:31.665 -> incoming: /devices/shit_meter-898da8/config:
16:17:31.665 -> {"duty_cycle_interval_sec":1800}
16:17:31.665 -> No more task for GCloud, can go asleep
16:17:31.764 -> Telemetry: {"battery":3273,"temperature":2453,"distance":220}
16:17:32.063 -> Publishing telemetry... done
16:17:32.063 -> State: {"device_id":"shit_meter-898da8","chip_model":"ESP32-D0WDQ6","chip_revision":1,"chip_cores":2,"firmware_version":"0.0.3","ssid":"wl-500gpv2","bssid":"14:CC:20:62:C9:8F"}
16:17:32.063 -> Publishing state... done
16:17:32.063 -> State changed from: 3 to 4
16:17:32.561 -> Close connection to cloud... done
16:17:32.727 -> System going to sleep for 1800 seconds

Log of cloud:
[image]

sacculus, 2021-07-13:
Hi
As solution I added some additional time before put device asleep for (as I suppose) internal process can finish its job. After that described errors no more occurred.
So I close issue.

closed 2021-07-13 by sacculus