GoogleCloudPlatform/google-cloud-iot-arduino #227: ESp32-S2 compatibility?

lloydrichards, 2021-06-26:
Hey, and firstly thanks for all the wonderful work on this library 👍

I'm in the process of replacing my REST API for receiving IoT data with the Google Cloud IOT Core.  I have some simple sensors and am just dumping it into pubsub to be dealt with by BigQuery and Firebase. I've spent the morning trying to get the google-cloud-iot-arduino example to work with my latest board using an ESP32-S2 and am running into some issues with the Certificate.

Right now I'm just using the ESP32-mqtt example to get things working and am getting this message when the mqtt->connect()

`Connecting to MQTT
checking wifi...
Wifi connected.
Connecting...
Refreshing JWT
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/mimir-app-dev/locations/europe-west1/registries/mimir-iot-registry/devices/mimirOpen-01
Waiting 60 seconds, retry will likely fail
`

As far as I can tell all the configs are correct to my IoT Core so my ideas are either that i'v messed up and need to start taking the system apart, piece by piece.  or maybe the library isn't compatible with the ESP32-S2 or requires some tweaking.  I thought I would check first before spending the whole weekend on this only to find out I need to go back to my REST APi.  much appreciate any feedback!

vamshikrishnaoruganti, 2022-04-07:
could you please post your code?

lloydrichards, 2022-04-07:
I recently got this working on some older ESP32 models.  I e not opened the repo her as I still need to clean out some private keys from older versions.  Some examples though that really helped were from @alvarowolfx and his repo I'm IoT Core:

https://github.com/alvarowolfx/gcloud-iot-core-device-provisioning
https://github.com/alvarowolfx/google-cloud-iot-arduino

Hope those help and once I've got my repo in order i can share it here