GoogleCloudPlatform/google-cloud-iot-arduino #228: google-cloud-iot-arduino- ESP32

dgupt123, 2021-06-28:
Refreshing JWT not connected Settings incorrect or missing a cyper for SSL Connect with mqtt.2030.ltsapis.goog:8883 ClientId: projects/esp32-317614/locations/us-central1/registries/ESP32/devices/esp32-device Waiting 60 seconds, retry will likely fail

danielsigl, 2021-08-01:
Me too no, with every CA.

dgupt123, 2021-08-05:
> Me too, with every CA.

Yes, I tried a lot, Not able to figure out the issue.

danielsigl, 2021-08-05:
> > Me too, with every CA.
> 
> Yes, I tried a lot, Not able to figure out the issue.

For mee it worked like described here: https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/221#issuecomment-842392002

dgupt123, 2021-08-06:
> > > Me too, with every CA.
> > 
> > 
> > Yes, I tried a lot, Not able to figure out the issue.
> 
> For me it worked like described here: [#221 (comment)](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/issues/221#issuecomment-842392002)

I have added same code in my code, but I am getting same error, Attached pic
[image]