GoogleCloudPlatform/google-cloud-iot-arduino #230: Arduino mkr1000 GCP IoT Core and MQTT connect failure

netskink, 2021-07-21:
Hello, I'm trying to get an arduino mkr1000 board connected to IOT core via MQTT.  

I'm following the guide described in this repo (GoogleCloudPlatform/google-cloud-iot-arduino) README.md.  When I get to the end of the guide where the device attempts to connect to GCP, it fails with the following messages in the Serial console

```
Refreshing JWT
not connected
Settings incorrect or missing a cyper for SSL
Connect with mqtt.googleapis.com:8883
ClientId: projects/a-test-project/locations/us-central1/registries/MKR_GCP_Tutorial/devices/CESMKR1000
Waiting 60 seconds, retry will likely fail
```

## Here is a summary of what I did according to the README and my results.

### Hardware/Software Environment
Host Computer is Debian x86_64, v10.10
Arduino IDE 1.8.13

## Comments Regarding Quickstart procedure 
* I've tried various mqtt samples and libraries and "Google Cloud IoT JWT" library was already installed.
* GCP has a IoT Core Registry without any authtencation keys added as per the guide
* GCP IoT Core Registry has just MQTT protocol enabled.  Initially I had both.  An issue in this repo has a comment where someone mentioned enabling both MQTT and HTTP was a problem.  Enabling both does not seem to effect my problem.  In fails either way.
* I generated the .pem files according to the Eliptic Curve procedure.  I uploaded only the public pem to the device I created in GCP as described.  I removed any previous keys from older attempts.  In GCP the device authentication section lists one key with the following properties
    * key format E256
    * key value  its not shown.  It shows ****_****
    * expiration time is not specified.  Its shows -
*   Since I am using an older Arduino MKR1000, I used the file->examples->google cloud iot core jwt -> mkr1000-lwmqtt example.
* I extracted the private key bytes as described using this command `openssl ec -in ec_private.pem -noout -text
` and updated the example sketch ciotc_config.h accordingly.  I also modified this file for wifi, gcp settings. 
* In this git repo's README.md "Notes on the certificate" section, it specifies if you are using the MKR1000, then you need to add SSL cert to the MKR1000.  A link is provided to [hackster.io] (https://www.hackster.io/arichetta/add-ssl-certificates-to-mkr1000-93c89d).  I used this  process to run a firmware updater program on the MKR1000 and then use another program to download the cert from google.  I was able to run the program to fetch the cert and update the MKR1000 while it was running the firmware update program described on that page.  No warnings or errors.
* There is a following section after above in this projects README.md.  It has a section on "Create Registry keys" and a "Create Eliptic device keys".  I did not do this.   There is an earlier section on the device key which I did use.  The earlier section seemed to imply that registry keys were not needed.  I'm not sure what this section is for or why its specified.  I've used other devices previously without a registry key and the gcp help docs says its not required.  Regardless, I added a CA Cert as shown to see if that fixed my problem.  It did not.

Any advice is appreciated.

netskink, 2021-07-21:
I should also note that in GCP for the device, clicking the "view logs" link pulls up the logs but nothing is recorded.  I have also changed default log level to Debug.  Ie. Registry details says that cloud logging is "Debug".  I thought this would show connection errors.  Even when I was using other samples, nothing is ever shown on the gcp logging.

netskink, 2021-07-21:
I should also note, the README.md mentions jwt.io website.  I was unfamiliar with this site, but inserting a print of the jwt in the getJWT() routine, I get a jwt string which I can paste at the jwt.io website.  It decodes the jwt and lists the following:

## Headers
```
{
  "alg": "ES256",
  "typ": "JWT"
}
```
## payload/data
```
{
  "iat": 1626882243,
  "exp": 1626885843,
  "aud": "a-test-project"
}
```

Regarding this last part, the payload, it has an entry for EXP.  I am not sure what these fields are, but EXP seems like expiration.  I did not specify a key having a time limit, so I am not sure if this is an error or not.

netskink, 2021-07-23:
FWIW, I tried to use wireshark to diagnose this problem.  Here is a write up on that effort. 
https://gist.github.com/netskink/4f59d0d91944e85229947889bdb6911c

mrSilkie, 2021-08-03:
I have the same issue. I think there is a fundamental flaw in the instructions since far too many people are getting this issue.

netskink, 2021-08-08:
I've got another sample im using also.  I am trying to figure it out.  I recently made a wireshark capture file.  However its with a sample that uses the mkr1000 crypto's chip.  If you will, please take a look at this capture.  It might be related to this source.  I'll make a capture of this one soon.  I did full markdown with screenshots so its more than the giest above.

https://github.com/netskink/ssl_publy/blob/main/capture_dump.md

netskink, 2021-08-31:
I got this working with arduino example which uses crypto chip and this sample which does the crypto in software.  The github repo above shows how I got it working.

closed 2021-08-31 by netskink

cragona, 2021-11-05:
> I got this working with arduino example which uses crypto chip and this sample which does the crypto in software. The github repo above shows how I got it working.

have you found a viable solution for mkr1000?