GoogleCloudPlatform/google-cloud-iot-arduino #232: Project instructions unclear. How do download minimal root CA set to fix cypher error?

mrSilkie, 2021-08-03:
This past week has sucked on the esp32 front. I have not been able to get this project running and I think it's not due to the github project itself but rather poor documentation.

```
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1
Settings incorrect or missing a cyper for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/growsimple/locations/asia-east1/registries/grow-simple-nodes/devices/testNode
Waiting 60 seconds, retry will likely fail
```

Is an error that I keep bumping into. on githubmemory you'll quickly find that almost everybody is having this error.

https://githubmemory.com/repo/GoogleCloudPlatform/google-cloud-iot-arduino/issues

However there is one user who has a fix.

https://githubmemory.com/repo/GoogleCloudPlatform/google-cloud-iot-arduino/issues/221

Has an OP which claims that their fix to this error was to 

> 
> I was able to connect to Cloud IOT Core while using root certificate by following these steps:
> 
> Download primary and secondary crt files from the following link: (https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#downloading_mqtt_server_certificates)
> 
> Convert these '.crt' files in '.pem' file using the following command
> 
> openssl x509 -inform DER -in gtsltsr.crt -out primary.pem -text
> openssl x509 -inform DER -in GSR4.crt -out secondary.pem -text
> 
> And then copying the content from both these file which end up looking something like this in ciotc_config.h file

Which sounds easy until you visit the google docs and they are not written for embedded developers. 

https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#downloading_mqtt_server_certificates

> To use TLS transport, devices must verify Cloud IoT Core server certificates to ensure they're communicating with Cloud IoT Core rather than an impersonator. The following certificate packages support verification:
> 
>     The complete Google root CA certification package (128 KB) for mqtt.googleapis.com.
>     Google's minimal root CA set (<1 KB) for mqtt.2030.ltsapis.goog. The minimal root CA set includes a primary and backup certificate.
> 
> After downloading Google root CA certificates to your device, you can configure an MQTT client to authenticate the device, connect to the MQTT server, and communicate over the MQTT bridge.

Which is chock full of information on how / where to actually download the .crt file. I am stuck. The rabbit hole comes to a dead end. I've already tried copying and pasting the results of openssl s_client -showcerts -connect mqtt.2030.ltsapis.goog:8883 with no success.

mrSilkie, 2021-08-03:
I believe I have figured out how to download the root certificate needed. via

wget pki.goog/roots.pem

Still after copying and pasting the contents over to my ciotc_config file i get no positive results. I've seen content that makes this certificate process easy in the arudino IDE. Is there not something available for Platform.IO?

netskink, 2021-08-10:
I don''t know platformIO, but the way to load that according to an arduino blog/tutorial (sorry I don't remember the url) is to use the firmware/cerficate loader utility.

I specified only a single domain for the certificate you refer above as:

pki.goog/:443

duongnghiephuy, 2021-11-08:
For anyone facing the same issue "Settings incorrect or missing a cyper for SSL", this might help. My error was because root_cert was not used anywhere in the connection. To fix that, we can

Change `netClient` class to `WiFiClientSecure` 
- In `esp32-mqtt.h` (line 36)
- In `CloudIoTCoreMqtt.cpp` (line 27) 
- In `CloudIoTMqtt.h` (line 36, 40) 

Then `#include <WiFiClientSecure.h>` in `CloudIoTMqtt.h`. 
In `esp32-mqtt.h`, set certificate 
```
void setupCloudIoT()
{
  device = new CloudIoTCoreDevice(
      project_id, location, registry_id, device_id,
      private_key_str);

  setupWifi();
  netClient = new WiFiClientSecure();
  netClient->setCACert(root_cert); //  Set certificate

  mqttClient = new MQTTClient(512);
  mqttClient->setOptions(180, true, 1000); // keepAlive, cleanSession, timeout
  mqtt = new CloudIoTCoreMqtt(mqttClient, netClient, device);
  mqtt->setUseLts(true);
  mqtt->startMQTT();
}
```

Download [primary cert](https://pki.goog/gtsltsr/gtsltsr.crt) and [backup cert](https://pki.goog/gsr4/GSR4.crt)  (Google's minimal root CA set (<1 KB) for mqtt.2030.ltsapis.goog)  then convert as mentioned above before copying them to `root_cert` in `ciotc_config.h`. Both certs are necessary as Google Cloud might switch without warning. 
Or just copy the cert below to `root_cert` in `ciotc_config.h`.

```
const char *root_cert =
    "-----BEGIN CERTIFICATE-----\n"
    "MIIBxTCCAWugAwIBAgINAfD3nVndblD3QnNxUDAKBggqhkjOPQQDAjBEMQswCQYD\n"
    "VQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZpY2VzIExMQzERMA8G\n"
    "A1UEAxMIR1RTIExUU1IwHhcNMTgxMTAxMDAwMDQyWhcNNDIxMTAxMDAwMDQyWjBE\n"
    "MQswCQYDVQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZpY2VzIExM\n"
    "QzERMA8GA1UEAxMIR1RTIExUU1IwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAATN\n"
    "8YyO2u+yCQoZdwAkUNv5c3dokfULfrA6QJgFV2XMuENtQZIG5HUOS6jFn8f0ySlV\n"
    "eORCxqFyjDJyRn86d+Iko0IwQDAOBgNVHQ8BAf8EBAMCAYYwDwYDVR0TAQH/BAUw\n"
    "AwEB/zAdBgNVHQ4EFgQUPv7/zFLrvzQ+PfNA0OQlsV+4u1IwCgYIKoZIzj0EAwID\n"
    "SAAwRQIhAPKuf/VtBHqGw3TUwUIq7TfaExp3bH7bjCBmVXJupT9FAiBr0SmCtsuk\n"
    "miGgpajjf/gFigGM34F9021bCWs1MbL0SA==\n"
    "-----END CERTIFICATE-----\n"
    "-----BEGIN CERTIFICATE-----\n"
    "MIIB4TCCAYegAwIBAgIRKjikHJYKBN5CsiilC+g0mAIwCgYIKoZIzj0EAwIwUDEk\n"
    "MCIGA1UECxMbR2xvYmFsU2lnbiBFQ0MgUm9vdCBDQSAtIFI0MRMwEQYDVQQKEwpH\n"
    "bG9iYWxTaWduMRMwEQYDVQQDEwpHbG9iYWxTaWduMB4XDTEyMTExMzAwMDAwMFoX\n"
    "DTM4MDExOTAzMTQwN1owUDEkMCIGA1UECxMbR2xvYmFsU2lnbiBFQ0MgUm9vdCBD\n"
    "QSAtIFI0MRMwEQYDVQQKEwpHbG9iYWxTaWduMRMwEQYDVQQDEwpHbG9iYWxTaWdu\n"
    "MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEuMZ5049sJQ6fLjkZHAOkrprlOQcJ\n"
    "FspjsbmG+IpXwVfOQvpzofdlQv8ewQCybnMO/8ch5RikqtlxP6jUuc6MHaNCMEAw\n"
    "DgYDVR0PAQH/BAQDAgEGMA8GA1UdEwEB/wQFMAMBAf8wHQYDVR0OBBYEFFSwe61F\n"
    "uOJAf/sKbvu+M8k8o4TVMAoGCCqGSM49BAMCA0gAMEUCIQDckqGgE6bPA7DmxCGX\n"
    "kPoUVy0D7O48027KqGx2vKLeuwIgJ6iFJzWbVsaj8kfSt24bAgAXqmemFZHe+pTs\n"
    "ewv4n4Q=\n"
    "-----END CERTIFICATE-----\n";
```

I also did simple visualization of data sent using Cloud Function and InFluxDB 
[here](https://github.com/duongnghiephuy/Simple_ESP32_Sensor_GoogleCloud)

netskink, 2021-11-09:
FWIW, I'm giving a talk this saturday on SSL and how it relates to this sample and the similar Arduinio sample.  Here is a link to my presentation https://docs.google.com/presentation/d/11Qe8t6ijRLvJmTuNms9nt65N3AZyAZNXwVhpvgmxCyg/edit?usp=sharing

Here is a link to the event on saturday https://gdg.community.dev/events/details/google-gdg-cloud-rtp-presents-southern-region-devfest/

ChaserVasya, 2021-11-13:
I have solved it by increasing timeout from example`s 1000 to 10000.