GoogleCloudPlatform/google-cloud-iot-arduino #241: Settings incorrect or missing a cyper for SSL

simogaspa84, 2022-02-21:
Hi.. 
i am trying to use the library for the chip ESP32 under the Arduino Framework (Platformio).
I am always getting this infofrom the debug

Settings incorrect or missing a cyper for SSL

Any idea or experience on that?

Thanks

Embedded-Solution, 2022-03-01:
Hi, 
Experiencing same issue using example [Esp32-lwmqtt](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Esp32-lwmqtt) with last commit https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/commit/2e7579019aa3192201dbdf50c3f76b60457c36f1

I'm using ESP32 the Google's minimal root CA. 
```
Starting wifi
Connecting to WiFi
Waiting on time sync...
checking wifi...Connecting...
Refreshing JWT
not connected
Settings incorrect or missing a cypher for SSL
Connect with mqtt.2030.ltsapis.goog:8883
ClientId: projects/{my-project-id}/locations/europe-west1/registries/{my-registry}/devices/{my-device}
Waiting 60 seconds, retry will likely fail
```

Any idea how to solve this issue? 

Thanks

robocircuits, 2022-03-10:
Same issue with me

simogaspa84, 2022-03-10:
Guys you have to use the google certificate for connecting with endpoint mqtt.2030.ltsapis.goog.
You have to use primary and backup certificate together ..  
The format is the following 

[image]

prakharmaurya, 2022-03-18:
may be you didn't saw last commit -> "[Fix cypher for SSL error in ESP32-lwmqtt example](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/commit/2e7579019aa3192201dbdf50c3f76b60457c36f1)", do the setCACert thing that resolved for me.

kargarisaac, 2022-05-11:
It was working for me, but today suddenly stopped working. I cannot program the board to connect to iot core and get the same error as above. Should we update the `root_cert` regularly?! 

I have the following in my code which is based on the commit:
```C++

const char *root_cert =
    "-----BEGIN CERTIFICATE-----\n"
    "MIIBxTCCAWugAwIBAgINAfD3nVndblD3QnNxUDAKBggqhkjOPQQDAjBEMQswCQYD\n"
    "VQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZpY2VzIExMQzERMA8G\n"
    "A1UEAxMIR1RTIExUU1IwHhcNMTgxMTAxMDAwMDQyWhcNNDIxMTAxMDAwMDQyWjBE\n"
    "MQswCQYDVQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZpY2VzIExM\n"
    "QzERMA8GA1UEAxMIR1RTIExUU1IwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAATN\n"
    "8YyO2u+yCQoZdwAkUNv5c3dokfULfrA6QJgFV2XMuENtQZIG5HUOS6jFn8f0ySlV\n"
    "eORCxqFyjDJyRn86d+Iko0IwQDAOBgNVHQ8BAf8EBAMCAYYwDwYDVR0TAQH/BAUw\n"
    "AwEB/zAdBgNVHQ4EFgQUPv7/zFLrvzQ+PfNA0OQlsV+4u1IwCgYIKoZIzj0EAwID\n"
    "SAAwRQIhAPKuf/VtBHqGw3TUwUIq7TfaExp3bH7bjCBmVXJupT9FAiBr0SmCtsuk\n"
    "miGgpajjf/gFigGM34F9021bCWs1MbL0SA==\n"
    "-----END CERTIFICATE-----\n";
```

Also, I see the token in the output of the following command. It is not the first but somewhere in between.

When I use ` openssl s_client -showcerts -connect mqtt.2030.ltsapis.goog:8883`, I see the following output:

```
CONNECTED(0000018C)
depth=1 C = US, O = Google Trust Services LLC, CN = GTS LTSX
verify error:num=20:unable to get local issuer certificate
verify return:1
depth=0 CN = *.2030.ltsapis.goog
verify return:1
---
Certificate chain
 0 s:CN = *.2030.ltsapis.goog
   i:C = US, O = Google Trust Services LLC, CN = GTS LTSX
-----BEGIN CERTIFICATE-----
MIICvTCCAmOgAwIBAgIUfXAmIec7M7FjzbdZBZjU0wbRPR8wCgYIKoZIzj0EAwIw
RDELMAkGA1UEBhMCVVMxIjAgBgNVBAoTGUdvb2dsZSBUcnVzdCBTZXJ2aWNlcyBM
TEMxETAPBgNVBAMTCEdUUyBMVFNYMB4XDTIyMDMyOTAwMDAwMFoXDTIzMDMyODAw
MDAwMFowHjEcMBoGA1UEAwwTKi4yMDMwLmx0c2FwaXMuZ29vZzBZMBMGByqGSM49
AgEGCCqGSM49AwEHA0IABJb3OTDbhSah6bxSnBE1GIN8fdcKnyaMkOWym8MC/DYb
xn8PO16tuykiIAfyudGV/clMSSD7r50wBCzwmL3/+G+jggFXMIIBUzATBgNVHSUE
DDAKBggrBgEFBQcDATAOBgNVHQ8BAf8EBAMCB4AwHgYDVR0RBBcwFYITKi4yMDMw
Lmx0c2FwaXMuZ29vZzAMBgNVHRMBAf8EAjAAMB8GA1UdIwQYMBaAFLMrq6BIHH4T
iskIxEBCIqI2UeIKMGkGCCsGAQUFBwEBBF0wWzAvBggrBgEFBQcwAoYjaHR0cDov
L3BraS5nb29nL2d0c2x0c3IvZ3RzbHRzeC5jcnQwKAYIKwYBBQUHMAGGHGh0dHA6
Ly9vY3NwLnBraS5nb29nL0dUU0xUU1gwIQYDVR0gBBowGDAMBgorBgEEAdZ5AgUD
MAgGBmeBDAECATAwBgNVHR8EKTAnMCWgI6Ahhh9odHRwOi8vY3JsLnBraS5nb29n
L0dUU0xUU1guY3JsMB0GA1UdDgQWBBTjW5lTveo1QiBCQ7jslByIITAlUTAKBggq
hkjOPQQDAgNIADBFAiBn/6GkEkYTej1fYmG/qFRlm9HvuhEAKfTKPvXvCfCufwIh
AL/M3xuaDoWqbliWgeH5Ig+JjUT1Kw8P1Jh/EIcfAFuZ
-----END CERTIFICATE-----
 1 s:C = US, O = Google Trust Services LLC, CN = GTS LTSX
   i:C = US, O = Google Trust Services LLC, CN = GTS LTSR
-----BEGIN CERTIFICATE-----
MIIC0TCCAnagAwIBAgINAfQKmcm3qFVwT0+3nTAKBggqhkjOPQQDAjBEMQswCQYD
VQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZpY2VzIExMQzERMA8G
A1UEAxMIR1RTIExUU1IwHhcNMTkwMTIzMDAwMDQyWhcNMjkwNDAxMDAwMDQyWjBE
MQswCQYDVQQGEwJVUzEiMCAGA1UEChMZR29vZ2xlIFRydXN0IFNlcnZpY2VzIExM
QzERMA8GA1UEAxMIR1RTIExUU1gwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAARr
6/PTsGoOg9fXhJkj3CAk6C6DxHPnZ1I+ER40vEe290xgTp0gVplokojbN3pFx07f
zYGYAX5EK7gDQYuhpQGIo4IBSzCCAUcwDgYDVR0PAQH/BAQDAgGGMB0GA1UdJQQW
MBQGCCsGAQUFBwMBBggrBgEFBQcDAjASBgNVHRMBAf8ECDAGAQH/AgEAMB0GA1Ud
DgQWBBSzK6ugSBx+E4rJCMRAQiKiNlHiCjAfBgNVHSMEGDAWgBQ+/v/MUuu/ND49
80DQ5CWxX7i7UjBpBggrBgEFBQcBAQRdMFswKAYIKwYBBQUHMAGGHGh0dHA6Ly9v
Y3NwLnBraS5nb29nL2d0c2x0c3IwLwYIKwYBBQUHMAKGI2h0dHA6Ly9wa2kuZ29v
Zy9ndHNsdHNyL2d0c2x0c3IuY3J0MDgGA1UdHwQxMC8wLaAroCmGJ2h0dHA6Ly9j
cmwucGtpLmdvb2cvZ3RzbHRzci9ndHNsdHNyLmNybDAdBgNVHSAEFjAUMAgGBmeB
DAECATAIBgZngQwBAgIwCgYIKoZIzj0EAwIDSQAwRgIhAPWeg2v4yeimG+lzmZAC
DJOlalpsiwJR0VOeapY8/7aQAiEAiwRsSQXUmfVUW+N643GgvuMH70o2Agz8w67f
SX+k+Lc=
-----END CERTIFICATE-----
---
Server certificate
subject=CN = *.2030.ltsapis.goog

issuer=C = US, O = Google Trust Services LLC, CN = GTS LTSX

---
No client certificate CA names sent
Requested Signature Algorithms: ECDSA+SHA256:RSA-PSS+SHA256:RSA+SHA256:ECDSA+SHA384:RSA-PSS+SHA384:RSA+SHA384:RSA-PSS+SHA512:RSA+SHA512:RSA+SHA1
Shared Requested Signature Algorithms: ECDSA+SHA256:RSA-PSS+SHA256:RSA+SHA256:ECDSA+SHA384:RSA-PSS+SHA384:RSA+SHA384:RSA-PSS+SHA512:RSA+SHA512
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 1771 bytes and written 434 bytes
Verification error: unable to get local issuer certificate
---
New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
Server public key is 256 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 20 (unable to get local issuer certificate)
---
read:errno=0
```

Can someone explain how it works? do we need to update it regularly? What about a device that goes to production?

Thank you

kargarisaac, 2022-05-11:
I also see the following in my serial outputs (not sure if I ha it before or not):
```
14:03:50.484 -> E (38263) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
14:03:50.484 -> E (38263) task_wdt:  - IDLE (CPU 0)
14:03:50.484 -> E (38263) task_wdt: Tasks currently running:
14:03:50.484 -> E (38263) task_wdt: CPU 0: loopTask
```

juniorsaldanha, 2022-08-13:
Does any have found an solution? I'm trying to use the minimal root which is not working, and the lts one gives me multiples certificates , should I use all of them, how? in the same const char ?

woodalgo, 2022-11-07:
.. I have been encountering this weird error.. I think it is something to do with the encoding of the JWT , if you have a non alphanumeric character in your device ID , the encoding will more often than not throw up a character that cause the jwt to be malformed.. rename your device id , and it might work.