GoogleCloudPlatform/google-cloud-iot-arduino #242: Google library and ethernet shield Enc28J60

simogaspa84, 2022-03-31:
Hi .. 

I would like to use your library instead that with the wifi network using an ethernet cable.
The Ethernet shield is Enc28J60...
Can you provide some code examples about where modify and where to switch from wifi to the ethernet shield..?
Thanks a lot

gguuss, 2022-04-11:
I am not actively contributing to or maintaining the library as I'm no longer at Google. You may want to look at the [MKR1000 example](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/MKR1000-lwmqtt/mkr1000-mqtt.h) because the Arduino WiFiSSLClient library shares capabilities with their ethernet library and you may be able to port the MKR example to [Arduino Ethernet](https://www.arduino.cc/en/reference/ethernet).

simogaspa84, 2022-04-12:
Hi @gguuss thanks for your reply... 
We have all the boards with ethernet shield Enc28J60.
How feasible is in your opinion starting from MKR1000 example and make it work for shield Enc28J60 ?
Do you know someone that I can contact to do this job (paying for sure)
Currently I am using this lib for working with shield Enc28J60.

https://github.com/simogaspa84/EthernetENC

Thanks again for your kind support