GoogleCloudPlatform/google-cloud-iot-arduino #243: Esp32-c3 support?

kargarisaac, 2022-04-02:
Hello,

I'm trying to use the esp32-lwmqtt example with new esp32-c3 in arduino but it doesn't work (I'm not a C guy and rarely code in C++ - was using mongoose os with esp32 but it does not support my new board, esp32-c3, and I have to use Arduino, and if it doesn't work I have to go for C). At the beginning it was a conflict between `Wifi.h` from esp32 and Arduino and by deleting the wifi library of Arduino, now I get the following error:

```
c:/users/isaac/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\Google_Cloud_IoT_Core_JWT\CloudIoTCoreMqtt.cpp.o: in function `CloudIoTCoreMqtt::startMQTTAdvanced()':
C:\Users\Isaac\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src/CloudIoTCoreMqtt.cpp:168: undefined reference to `messageReceivedAdvanced(MQTTClient*, char*, char*, int)'
c:/users/isaac/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\Isaac\Documents\Arduino\libraries\Google_Cloud_IoT_Core_JWT\src/CloudIoTCoreMqtt.cpp:169: undefined reference to `messageReceivedAdvanced(MQTTClient*, char*, char*, int)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32C3 Dev Module.
```

Is it because of not supporting the board? I can run some arduino examples on the board but not sure if this error is because of the cloud-iot-arduino not supporting the board or something else.

svdrummer, 2022-04-03:
Have you set the required information in the ciotc_config.h file?
Have a look in the example folder. There are three files to this project.
Only one is an .ino folder

kargarisaac, 2022-04-04:
Thank you for your answer. I use the sample and edit configs. I can connect to google cloud iot core and send data when I use regular esp32 devkit but when I use the new board with esp32-c3, I get the above error.

This is the whole error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32C3 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.cpp:12: warning: "assert" redefined
 #define assert(condition) do{} while(0)
 
In file included from c:\users\isaac\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\sys-include\sys\reent.h:503,
                 from C:\Users\Isaac\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/newlib/platform_include/sys/reent.h:17,
                 from c:\users\isaac\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\sys-include\string.h:11,
                 from c:\users\isaac\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\include\c++\8.4.0\cstring:42,
                 from C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.h:23,
                 from C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.cpp:8:
C:\Users\Isaac\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/newlib/platform_include/assert.h:50: note: this is the location of the previous definition
 #define assert(__e) (__builtin_expect(!!(__e), 1) ? (void)0 : __assert_func (__FILENAME__, __LINE__, \
 
c:/users/isaac/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\google-cloud-iot-arduino-master\CloudIoTCoreMqtt.cpp.o: in function `CloudIoTCoreMqtt::startMQTTAdvanced()':
C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src/CloudIoTCoreMqtt.cpp:168: undefined reference to `messageReceivedAdvanced(MQTTClient*, char*, char*, int)'
c:/users/isaac/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src/CloudIoTCoreMqtt.cpp:169: undefined reference to `messageReceivedAdvanced(MQTTClient*, char*, char*, int)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32C3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

I can also use the board with normal examples such as reading chip info and `ESP32DevelopmentBoard` form `MQTT` module. It seems a problem with gcc in the error for iot core library but I'm not an expert in C/C++. Any help would be appreciated.

Thank you

kargarisaac, 2022-04-04:
The problem is solved now. I defined `messageReceivedAdvanced()` function in my code and the error is gone. After that another problem appeard that the board was going into boot loop that by changing the gpio pin from 13 to 19 (line 32 in `ciotc_config.h`), that problem is solved too.

closed 2022-04-04 by kargarisaac

matteocordray, 2022-04-15:
I'm having this same issue, where did you define the `messageReceivedAdvanced()` function in your code?

kargarisaac, 2022-04-15:
just below `messageReceived()` function in [this](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/examples/Esp32-lwmqtt/esp32-mqtt.h) file.

sth like:
```
void messageReceived(String &topic, String &payload){
  Serial.println("incoming: " + topic + " - " + payload);
}
```

matteocordray, 2022-04-15:
This is what I put and I'm still getting the same error :(
```
// The MQTT callback function for commands and configuration updates
// Place your message handler code here.
void messageReceived(String &topic, String &payload){
  Serial.println("incoming: " + topic + " - " + payload);
}

void messageReceivedAdvanced(MQTTClient *client, char topic[], char bytes[], int length);
```
Error:
```
AppData\Local\Temp\arduino_build_925918\libraries\Google_Cloud_IoT_Core_JWT\CloudIoTCoreMqtt.cpp.o: in function `CloudIoTCoreMqtt::startMQTTAdvanced()':
libraries\Google_Cloud_IoT_Core_JWT\src/CloudIoTCoreMqtt.cpp:168: undefined reference to `messageReceivedAdvanced(MQTTClient*, char*, char*, int)'
```

matteocordray, 2022-04-15:
Got it! Was dumb enough to not actually define the function...this should definitely be fixed in the main code with some `#ifdef ESP32_C3` since it seems it's only with this microcontroller. LED worked as well. Thank you!